bernhard-42/jupyter-cadquery #2: BUG,BLD: JupyterLab 1.0

westurner, 2019-07-09:
With JupyterLab 1.0 (released about week ago), I'm getting this error when I run the example in the README ( https://github.com/bernhard-42/jupyter-cadquery#b-example-cadquery-using-sidecar )

```
 Error displaying widget: model not found
```

It doesn't say which code is throwing this exception.
It looks like some changes were necessary to support JupyterLab 1.0:
- https://github.com/matplotlib/jupyter-matplotlib/issues/112#issuecomment-509472926

With this Makefile build script (on Fedora, though that shouldn't make any difference) that specifies `jupyterlab<1.0` I still get `Error displaying widget` in the sidecar.

```Makefile
_SRC=../
install-gui:
	conda install -y numpy 'jupyterlab<1.0' python=3.6 nodejs
	conda install -y -c conda-forge -c cadquery pyparsing pythonocc-core
	pip install --upgrade git+https://github.com/CadQuery/cadquery
	pip install ipywidgets pythreejs sidecar dataclasses
	jupyter labextension install @jupyter-widgets/jupyterlab-manager jupyter-threejs @jupyter-widgets/jupyterlab-sidecar
	#pip install -e git+https://github.com/bernhard-42/jupyter-cadquery.git#egg=jupyter_cadquery
	#cd '${_SRC}/jupyter-cadquery' && \
	#	jupyter-labextension install js
	cd '${_SRC}' && test -d jupyter-cadquery || git clone https://github.com/bernhard-42/jupyter-cadquery.git
	cd '${_SRC}/jupyter-cadquery' && pip install .
	cd '${_SRC}/jupyter-cadquery' && jupyter-labextension install js
```
### Workarounds
- Use the Docker container instead

westurner, 2019-07-09:
The JupyterLab changelog lists a bunch of things for extension development:
https://jupyterlab.readthedocs.io/en/stable/getting_started/changelog.html#extension-development-changes

westurner, 2019-07-09:
Actually, when I try to build the Dockerfile, I get version conflicts; presumably because versions aren't pinned. It may be easier to either and/or:

- Create an `environment.yml` (`conda env export -f environment.yml`)
  - Dockerfile: ``COPY environment.yml``
- Store the built docker image on docker hub with a version tag

```
Step 8/25 : RUN conda run -n pycq jupyter labextension install @jupyter-widgets/jupyterlab-manager jupyter-threejs @jupyter-widgets/jupyterlab-sidecar
 ---> Running in 633b57ffed1d

[... npm + webpack output ... ]

> /home/cq/.conda/envs/pycq/bin/npm pack @jupyter-widgets/jupyterlab-manager
Incompatible extension:

"@jupyter-widgets/jupyterlab-manager@1.0.0" is not compatible with the current JupyterLab
Conflicting Dependencies:
JupyterLab              Extension        Package
>=0.19.1 <0.20.0        >=1.0.0 <2.0.0   @jupyterlab/application
>=2.2.1 <3.0.0          >=3.0.0 <4.0.0   @jupyterlab/coreutils
>=0.19.2 <0.20.0        >=1.0.0 <2.0.0   @jupyterlab/notebook
>=0.19.1 <0.20.0        >=1.0.0 <2.0.0   @jupyterlab/rendermime
>=3.2.1 <4.0.0          >=4.0.0 <5.0.0   @jupyterlab/services

Found compatible version: 0.38.1
> /home/cq/.conda/envs/pycq/bin/npm pack @jupyter-widgets/jupyterlab-manager@0.38.1
Node v11.14.0

> /home/cq/.conda/envs/pycq/bin/npm pack jupyter-threejs
Node v11.14.0

> /home/cq/.conda/envs/pycq/bin/npm pack @jupyter-widgets/jupyterlab-sidecar
Incompatible extension:

"@jupyter-widgets/jupyterlab-sidecar@0.4.0" is not compatible with the current JupyterLab
Conflicting Dependencies:
JupyterLab              Extension        Package
>=0.19.1 <0.20.0        >=1.0.0 <2.0.0   @jupyterlab/application

Found compatible version: 0.3.0
> /home/cq/.conda/envs/pycq/bin/npm pack @jupyter-widgets/jupyterlab-sidecar@0.3.0
Node v11.14.0

> node /home/cq/.conda/envs/pycq/lib/python3.6/site-packages/jupyterlab/staging/yarn.js install
> node /home/cq/.conda/envs/pycq/lib/python3.6/site-packages/jupyterlab/staging/yarn.js run build
```

westurner, 2019-07-09:
It's not the sidecar widget; because -- with the Docker container built from the Dockerfile today -- when I comment this out, I still get:

```python
#from jupyter_cadquery import set_sidecar

#set_sidecar("CadQuery")  # force usage of one cad view on the right
```

Output:
```
Overwriting auto display for cadquery Workplane and Shape

Error displaying widget

[1]: <jupyter_cadquery.cad_display.CadqueryDisplay at 0x6ae20a5263c7>
```

westurner, 2019-07-09:
It's probably possible to create a test notebook and run that within the Docker container whenever pinned package versions change. A CI test script could run this either as a shell command, or with subprocess in a pytest test function:

```
jupyter nbconvert --execute test_notebook_001.ipynb
```

bernhard-42, 2019-07-09:
Thanks Wes for looking into it!

I am now building an environment.yml file with pinned versions and will also move that to the Dockerfile. Currently I am stuck with versions issues, but will push a new version as soon as it is resolved

westurner, 2019-07-10:
Hey thanks! You could probably COPY and call the postBuid script with the
Jupyter lab extension installs in it, too. With the COPY instruction,
docker (and/or buildah) will rebuild that and subsequent layers whenever
the copied file changes.

It's nice to have everything in the Dockerfile,
but DRY and thus maintainable to keep the packages and versions in the
environment.yml

bernhard-42, 2019-07-10:
@westurner I have made my way through the version jungle.
There is now a conda environment file for jupyterlab 0.35 (which includes jupyterlab-sidecar) and one for jupyterlab 1.0 (where we currently cannot install jupyterlab-sidecar)
If you want, you could give it a try and see whether it works for you.

I will continue to add the conde env files to docker (just looking as repo2docker)

bernhard-42, 2019-07-11:
@westurner 
I could solve the sidecar issue. Now all features are available for Jupyterlab 0.35 and 1.0

I have also changed the Dockerfile to COPY the conda environment file. It allows to build both a container for jupyterlab 0.35 and 1.0. Unfortunately, the docker images are pretty large:

```
REPOSITORY                          TAG            IMAGE ID            CREATED             SIZE
bernhard-42/jupyter-cadquery-0.35   0.9.1          0fd42259fe56        9 minutes ago       3.47GB
bernhard-42/jupyter-cadquery-1.0    0.9.1          ff5185949329        19 minutes ago      3.64GB
```

Additionally I have added my examples folder into the container so that a quick check is possible.

jmwright, 2019-07-11:
@bernhard-42 Is using the Alpine docker image an option? That should reduce the image size significantly.

bernhard-42, 2019-07-12:
@jmwright The miniconda3 base has 457MB, but the real huge layer is the conda environment. It has 2.82 GB. So alpine might save us maybe 200-300 MB, but that is not  a big win

bernhard-42, 2019-09-21:
Closing, since issue is resolved

closed 2019-09-21 by bernhard-42