bernhard-42/jupyter-cadquery #3: Error with: pip3 install .

Zawarud, 2019-08-09:
This is entire error code 👍 
`ERROR: Command errored out with exit status 1:
     command: 'c:\users\wolvi\.conda\envs\cd-jl-1.0\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\wolvi\\AppData\\Local\\Temp\\pip-req-build-2p204ejw\\setup.py'"'"'; __file__='"'"'C:\\Users\\wolvi\\AppData\\Local\\Temp\\pip-req-build-2p204ejw\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base pip-egg-info
         cwd: C:\Users\wolvi\AppData\Local\Temp\pip-req-build-2p204ejw\
    Complete output (113 lines):
    setup.py entered
    $PATH=c:\users\wolvi\.conda\envs\cd-jl-1.0\Library\bin;C:\Users\wolvi\.conda\envs\cd-jl-1.0;C:\Users\wolvi\.conda\envs\cd-jl-1.0\Library\mingw-w64\bin;C:\Users\wolvi\.conda\envs\cd-jl-1.0\Library\usr\bin;C:\Users\wolvi\.conda\envs\cd-jl-1.0\Library\bin;C:\Users\wolvi\.conda\envs\cd-jl-1.0\Scripts;C:\Users\wolvi\.conda\envs\cd-jl-1.0\bin;C:\ProgramData\Anaconda3\condabin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.1\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.1\libnvvp;C:\ProgramData\Anaconda3;C:\ProgramData\Anaconda3\Library\mingw-w64\bin;C:\ProgramData\Anaconda3\Library\usr\bin;C:\ProgramData\Anaconda3\Library\bin;C:\ProgramData\Anaconda3\Scripts;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\Program Files\MATLAB\R2018b\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\MATLAB\MATLAB Compiler Runtime\v711\runtime\win32;C:\Program Files\SlikSvn\bin;C:\Program Files\Git\cmd;C:\Program Files\CMake\bin;C:\Program Files\NVIDIA Corporation\Nsight Compute 2019.3.0;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Blender Foundation\Blender;C:\Program Files\Blender Foundation\Blender\2.80;C:\Program Files\Blender Foundation\Blender\2.80\python\lib;C:\Users\wolvi\AppData\Local\Programs\Python\Python37\Scripts;C:\Users\wolvi\AppData\Local\Programs\Python\Python37;C:\Tcl\bin;C:\Users\wolvi\AppData\Local\Microsoft\WindowsApps;.;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.1\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.1\libnvvp;C:\ProgramData\Anaconda3;C:\ProgramData\Anaconda3\Library\mingw-w64\bin;C:\ProgramData\Anaconda3\Library\usr\bin;C:\ProgramData\Anaconda3\Library\bin;C:\ProgramData\Anaconda3\Scripts;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\Program Files\MATLAB\R2018b\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\MATLAB\MATLAB Compiler Runtime\v711\runtime\win32;C:\Program Files\SlikSvn\bin;C:\Program Files\Git\cmd;C:\Program Files\CMake\bin;C:\Program Files\NVIDIA Corporation\Nsight Compute 2019.3.0;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Blender Foundation\Blender;C:\Program Files\Blender Foundation\Blender\2.80;C:\Program Files\Blender Foundation\Blender\2.80\python\lib;.
    running egg_info
    running jsdeps
    6.7.0
    6.7.0
    Installing build dependencies with npm.  This may take a while...
    npm WARN prepublish-on-install As of npm@5, `prepublish` scripts are deprecated.
    npm WARN prepublish-on-install Use `prepare` for build steps and `prepublishOnly` for upload-only.
    npm WARN prepublish-on-install See the deprecation note in `npm help scripts` for more information.

    > jupyter_cadquery@0.9.1 prepublish C:\Users\wolvi\AppData\Local\Temp\pip-req-build-2p204ejw\js
    > webpack

    Hash: ecd4dbb11fa791d6081122e280df66799edb341cb37887592548f5420ba7
    Version: webpack 4.39.1
    Child
        Hash: ecd4dbb11fa791d60811
        Time: 1082ms
        Built at: 2019-08-09 09:43:02
               Asset      Size  Chunks             Chunk Names
        extension.js  1.19 KiB       0  [emitted]  main
        Entrypoint main = extension.js
        [0] ./lib/extension.js 1.42 KiB {0} [built]

        WARNING in configuration
        The 'mode' option has not been set, webpack will fallback to 'production' for this value. Set 'mode' option to 'development' or 'production' to enable defaults for each environment.
        You can also set it to 'none' to disable any default behavior. Learn more: https://webpack.js.org/configuration/mode/
    Child
        Hash: 22e280df66799edb341c
        Time: 3798ms
        Built at: 2019-08-09 09:43:04
         2 assets
        Entrypoint main = index.js index.js.map
         [1] ./lib/index.js 895 bytes {0} [built]
         [2] ./lib/image_button.js 1.93 KiB {0} [built]
         [3] (webpack)/buildin/global.js 472 bytes {0} [built]
         [4] (webpack)/buildin/module.js 497 bytes {0} [built]
         [5] ./lib/tree_view.js 8.51 KiB {0} [built]
         [6] ./style/index.css 1.06 KiB {0} [built]
         [7] ./node_modules/css-loader/dist/cjs.js!./style/index.css 3.13 KiB {0} [built]
        [11] ./package.json 983 bytes {0} [built]
            + 4 hidden modules

        WARNING in configuration
        The 'mode' option has not been set, webpack will fallback to 'production' for this value. Set 'mode' option to 'development' or 'production' to enable defaults for each environment.
        You can also set it to 'none' to disable any default behavior. Learn more: https://webpack.js.org/configuration/mode/

        ERROR in ./lib/tree_view.js
        Module not found: Error: Can't resolve '@jupyter-widgets/controls' in 'C:\Users\wolvi\AppData\Local\Temp\pip-req-build-2p204ejw\js\lib'
         @ ./lib/tree_view.js 15:0-76 50:16-30 51:19-33 67:15-30 69:4-19
         @ ./lib/index.js

        ERROR in ./lib/image_button.js
        Module not found: Error: Can't resolve '@jupyter-widgets/controls' in 'C:\Users\wolvi\AppData\Local\Temp\pip-req-build-2p204ejw\js\lib'
         @ ./lib/image_button.js 16:0-19:35 27:23-34 28:19-30 41:22-32
         @ ./lib/index.js
    Child
        Hash: b37887592548f5420ba7
        Time: 3502ms
        Built at: 2019-08-09 09:43:04
         2 assets
        Entrypoint main = index.js index.js.map
        [1] ./lib/embed.js 1.04 KiB {0} [built]
        [2] ./lib/image_button.js 1.93 KiB {0} [built]
[... 27 lines cut ...]
      File "C:\Users\wolvi\AppData\Local\Temp\pip-req-build-2p204ejw\setup.py", line 177, in <module>
        setup(**setup_args)
      File "c:\users\wolvi\.conda\envs\cd-jl-1.0\lib\site-packages\setuptools\__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "c:\users\wolvi\.conda\envs\cd-jl-1.0\lib\distutils\core.py", line 148, in setup
        dist.run_commands()
      File "c:\users\wolvi\.conda\envs\cd-jl-1.0\lib\distutils\dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "c:\users\wolvi\.conda\envs\cd-jl-1.0\lib\distutils\dist.py", line 974, in run_command
        cmd_obj.run()
      File "C:\Users\wolvi\AppData\Local\Temp\pip-req-build-2p204ejw\setup.py", line 46, in run
        raise e
      File "C:\Users\wolvi\AppData\Local\Temp\pip-req-build-2p204ejw\setup.py", line 39, in run
        self.distribution.run_command('jsdeps')
      File "c:\users\wolvi\.conda\envs\cd-jl-1.0\lib\distutils\dist.py", line 974, in run_command
        cmd_obj.run()
      File "C:\Users\wolvi\AppData\Local\Temp\pip-req-build-2p204ejw\setup.py", line 111, in run
        check_call([npmName, 'install'], cwd=node_root, stdout=sys.stdout, stderr=sys.stderr)
      File "c:\users\wolvi\.conda\envs\cd-jl-1.0\lib\subprocess.py", line 311, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['npm.cmd', 'install']' returned non-zero exit status 2.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.`

My version of jupyterlab is 1.0 on windows 1.0 and python 3.7

bernhard-42, 2019-08-10:
@Zawarud Thanks for giving it a try.
It looks like on windows (or your version of npm) the dependency `@jupyter-widgets/controls` should be explicitly in `package.json` (it should be anyhow, but up to now neither on mac nor on linux I ran into it). I've added this dependency, however have no access to Windows and can't test it.
Maybe you could give it another try?
Thanks

Zawarud, 2019-08-11:
My version of npm is 6.7.0. This is the code inside package.json :
`{
  "name": "jupyter_cadquery",
  "version": "0.9.1",
  "description": "An extension to render cadquery objects in JupyterLab via pythreejs",
  "author": "Bernhard Walter",
  "main": "lib/index.js",
  "license": "MIT",
  "repository": {
    "type": "git",
    "url": "https://github.com/bernhard-42/jupyter-cadquery.git"
  },
  "keywords": [
    "jupyter",
    "widgets",
    "ipython",
    "ipywidgets",
    "jupyterlab-extension"
  ],
  "files": [
    "lib/**/*.js",
    "style/*.css",
    "dist/*.js"
  ],
  "scripts": {
    "clean": "rimraf dist/",
    "prepublish": "webpack",
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "devDependencies": {
    "rimraf": "^2.6.3",
    "webpack": "^4.35.3",
    "webpack-cli": "^3.3.5"
  },
  "dependencies": {
    "css-loader": "^2.1.1",
    "jquery": "^3.4.0",
    "lodash": "^4.17.4",
    "style-loader": "^0.23.1"
  },
  "jupyterlab": {
    "extension": "lib/labplugin"
  }
}
`
I try : pip install jquery . 
`Collecting jquery
  Using cached https://files.pythonhosted.org/packages/05/45/9a6d7ff770b1279c901c8dfca1f9a9d6c9822d75a2bad834d0e2ddd4f8cd/jquery-1.2.3.zip
    ERROR: Command errored out with exit status 1:
     command: 'c:\programdata\anaconda3\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\wolvi\\AppData\\Local\\Temp\\pip-install-helqqsds\\jquery\\setup.py'"'"'; __file__='"'"'C:\\Users\\wolvi\\AppData\\Local\\Temp\\pip-install-helqqsds\\jquery\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base pip-egg-info
         cwd: C:\Users\wolvi\AppData\Local\Temp\pip-install-helqqsds\jquery\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\wolvi\AppData\Local\Temp\pip-install-helqqsds\jquery\setup.py", line 2, in <module>
        from turbogears.finddata import find_package_data
    ModuleNotFoundError: No module named 'turbogears'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.`
How to solve this?

bernhard-42, 2019-08-11:
I've added a change yesterday, which I don't see in the `package.json` you've pasted. Maybe you need to pull the repo again, so that the dependency to `@jupyter-widgets/controls` is there.

I did another try and had an issue with jquery (I guess you had a similar one, since you tried to install jquery. However `jquery` is for javascript and using `pip install` wouldn't work as expected.
For me it turned out that there was some inconsistency with already downloaded node modules. So I deleted `node_modules` from my `jupyter_cadquery` and my `jupyter_cadquery/js` folder and afterwards it worked.

I have tested it with `node v10.16.0`, `npm 6.9.0` under linux and `node v10.15.3`, `npm 6.4.1` und macos and it worked. As mentioned, I unfortunately can't do Windows tests atm.:

```bash
cd jupyter-cadquery   # assuming repo was already cloned
git pull
rm -fr node_modules/
rm -fr js/node_modules/

conda env create -f ./environment-jl-1.0.yml -n cq-jl
conda activate cq-jl

jupyter labextension install @jupyter-widgets/jupyterlab-manager @jupyter-widgets/jupyterlab-sidecar

pip install .
jupyter-labextension install js
```

Zawarud, 2019-08-11:
Now its solved. Ty.
```
Successfully built jupyter-cadquery
Installing collected packages: jupyter-cadquery
Successfully installed jupyter-cadquery-0.9.1
```
Now i try tu use in jupyterlab. I try tu build a reinforced concrete frame. Stay Tuned.

Zawarud, 2019-08-11:
how to move packages for use cadquery in base(root) enviroment ?

bernhard-42, 2019-08-11:
I‘d recommend to change into the conda environment that the installer created. cadquery is already installed there.

bernhard-42, 2019-08-11:
@Zawarud I have added the missing steps to run *jupyter-cadquery* to the Readme:

```bash
conda activate $CONDA_ENV
jupyter lab
```

bernhard-42, 2019-10-27:
no further activity, so I close the issue.
Feel free to reopen in case of further questions

closed 2019-10-27 by bernhard-42