bernhard-42/jupyter-cadquery #4: Suggestion: make an example with Binder

Peque, 2019-09-20:
[Binder](https://mybinder.readthedocs.io/en/latest/index.html) allows you to run Jupyter Notebooks (or Jupyter Lab) on the cloud.

[mybinder.org](https://mybinder.org/) is a Binder free service, which allows you to run notebooks from any public repository very easily.

It would be great to have an example of `jupyter-cadquery` that works on mybinder.

It seems you can [use Dockerfiles with Binder](https://mybinder.readthedocs.io/en/latest/tutorials/dockerfile.html?highlight=dockerfile).

Here is an [example of a Binder Dockerfile which is based on a Jupyter DockerStacks image](https://github.com/binder-examples/jupyter-stacks/blob/master/Dockerfile). I guess a `jupyter-cadquery` Docker file could be based on the [`base-notebok`](https://github.com/jupyter/docker-stacks/blob/master/base-notebook/Dockerfile).

I think it would be a great way to let people try the extension (and CadQuery, too) without the need to setup any environment. Then, once they love it, they can do so locally. :smile:

Peque, 2019-09-21:
Made a proof of concept:

https://github.com/Peque/cadquery-binder

Jupyter Lab is not working (nothing displayed). I am probably missing something.

bernhard-42, 2019-09-21:
@Peque This is a great suggestion and I had thought about it but ran out of time.
Using your example is a great jumpstart and I will add something along you proof of concept

For lab, btw. you just missed to install the ipywidgets and sidecar labextensions:
```bash
jupyter-labextension install @jupyter-widgets/jupyterlab-manager @jupyter-widgets/jupyterlab-sidecar
```

bernhard-42, 2019-09-22:
@Peque jupyter-cadquery now runs on binder (see also in [README.md](https://github.com/bernhard-42/jupyter-cadquery/blob/master/README.md)):

https://mybinder.org/v2/gh/bernhard-42/jupyter-cadquery/master?urlpath=lab

closed 2019-10-27 by bernhard-42