bernhard-42/jupyter-cadquery #5: Integration with Sphinx

Peque, 2019-09-28:
I am considering integrating this with a Sphinx extension I made which allowed you to display 3D renders out of CadQuery code in your documentation:

- [Source code](https://github.com/Peque/sphinxcadquery)
- [Example of 3D render](https://bulebule.readthedocs.io/en/latest/building.html#mounting)

In order to do so, I need to embed the HTML code in the documentation. Is there any function that outputs the required HTML in this project?

I will have a look myself, but in case you read it before I do... It will be faster. :stuck_out_tongue_winking_eye:

bernhard-42, 2019-09-28:
@Peque The whole project is heavily based ipywidgets (https://github.com/jupyter-widgets/ipywidgets) and pythreejs (which is a widget, https://github.com/jupyter-widgets/pythreejs). The image buttons and the navigation tree are widgets written by me. I have no idea whether this can be easily embedded somewhere else, since widgets have a mechanism to allow communication between the frontend (javascript) and the backend (python).
I have never looked into embedding it somewhere else than jupyter, however there is: https://ipywidgets.readthedocs.io/en/latest/embedding.html

Peque, 2019-09-28:
Thanks, you saved my a lot of time. :heart:

Peque, 2019-09-29:
@bernhard-42 I have been working on this without much success. I managed to create some HTML code with ipywidgets' `embed_minimal_html()`, but would not show anything when displayed on the browser.

I guess you are not using any intermediate format to pass from CadQuery/Python to threejs/JS (like glTF), right? If you were, that would be awesome, and much easier to work with. :joy:

jmwright, 2019-09-29:
I think that the CQParts team did some work with glTF, but I'm not sure if that's helpful here.

https://github.com/cqparts/cqparts

https://github.com/cqparts/cqparts/blob/c57d29e280ff3f92c3ac11337d517b7ae1d789b9/tests/t_cqparts/test_codecs.py#L313

Peque, 2019-09-29:
@jmwright Thanks for the pointer. :+1: :blush:

To be honest, I did not consider CQParts because it is not compatible with CadQuery-OCC.

@fragmuffin Would it be possible to (easily) use CQParts' glTF exporter with the current CadQuery-OCC?

jmwright, 2019-09-29:
I think there's been some work done on OCC compatibility.

https://github.com/cqparts/cqparts/issues/144

Probably worth tagging @zignig here too.

cua-cua, 2019-11-15:
@bernhard-42, based on your [previous comment](https://github.com/bernhard-42/jupyter-cadquery/issues/5#issuecomment-536201220) , I managed to get the cadquery object (without image buttons and the navigation tree) working with  `embed_minimal_html()`. Basically the idea was to use the pythreejs infrastructure to start with the debugging of html transformation.
From your example [ b) Example: CadQuery using Sidecar)](https://github.com/bernhard-42/jupyter-cadquery#b-example-cadquery-using-sidecar) I continued with:

<pre><code class="python">from jupyter_cadquery.cad_display import CadqueryView

view = CadqueryView()
for shape in a1.collect_shapes():
    view.add_shape(shape["name"], shape["shape"], shape["color"])`
`renderer = view.render()
</code></pre>

With this, I got a pythreejs.core.Renderer object that I displayed on jupyter notebook with:
<pre><code class="python"> from IPython.display import display
display(renderer)
</code></pre>

[image]

With a renderer from pythreejs, I already had the input for embed_minimal_html method with:
<pre><code class="python">from ipywidgets import embed
embed.embed_minimal_html('export.html', views=renderer, title='Renderer')
</code></pre>

The thing is that when I opened the html generated file ('[export.html](https://pastebin.com/eCahTvnF)') with the navigator, I could not see anything and the console error was related to a " Class TreeModel" not found.
 
[image]

I removed by hand the reference to the Class TreeModel, and that worked. [export_modified](https://pastebin.com/a5d9T54V)

My questions are:
- Could be possible to generate an html without a reference to class TreeModel, so use a pythreejs object without TreeModel reference?
- Could be possible to generate a object 'Renderer' that includes also buttons and navigation tree so it matches with embed_minimal_html? I tried to modify jupyter-cadquery to match exactly the structure proposed on [widget-cookiecutter](https://github.com/jupyter-widgets/widget-cookiecutter 
) but I got stuck.

bernhard-42, 2019-11-16:
@cua-cua Sounds like a good idea and I will look into it. 
I am travelling for the next 6 days, so earliest would be next weekend to check how this can be achieved. I hope that point 1 (leaving out TreeModel) would be an easy job, but no idea about your second point.
In case you have time and find a solution for either of the topics before I am back at my computer, I am also happy to accept pull requests ;-)

cua-cua, 2019-11-18:
@bernhard-42 , I managed to remove the reference to the class TreeModel. Somehow, it appears if I include the IPython display code on the jupyter notebook:

<pre><code class="python">from jupyter_cadquery.cad_display import CadqueryView
 from IPython.display import display
display(renderer)
</code></pre>

Removing that, the HTML output is OK:
 
<pre><code class="python">a1.collect_shapes()
    view = CadqueryView()
    for shape in a1.collect_shapes():
        view.add_shape(shape["name"], shape["shape"], shape["color"])
    renderer = view.render()
embed.embed_minimal_html('export.html', views=renderer, title='Renderer')
</code></pre>

So, at least one point solved ^^.

Have a nice trip!

cua-cua, 2019-11-18:
@bernhard-42, I was wrong on the last comment. `display(renderer)` is not the code that generates the TreeModel reference on HTML output, but `show(a1, axes=True, grid=True, ortho=True, axes0=True)` from the example. Removing it, the HTML output works (with pythreejs object)

bernhard-42, 2019-11-22:
@cua-cua 
I have fixed the `TreeModel` issue.

A straight forward approach now is to use

```python
w = show(a1)
```

adapt the cad view as wanted (axis, viewpoint, transparency, ...) and then call 

```python
from ipywidgets.embed import embed_minimal_html
embed_minimal_html('export.html', views=[w.cq_view.renderer], title='Renderer')
```

Using `w.cq_view.renderer` this will save the exact state of the visible pythreejs view.

Of course, you can also call `w = show(a1, *params)` where `params` is the dict of show parameters you'd like to be used and then call the `embed_minimal_html` with `views=w.cq_view.renderer`

Notes:

1. If you use `sidecar`then you need to close it first:

    ```
    from jupyter_cadquery import cad_display
    cad_display.SIDECAR.close()
    ```

2. For some reason (that I don't understand), every export is in perspective mode. Somehow the CombinedCamera of pythreejs does not save the state as orthographic when the view is orthographic.

3. Buttons and treeview can be exported, however the interaction logic of the UI is implemented in Python. So the treeview and the buttons won't have any effect in an exported HTML page.

cua-cua, 2019-11-25:
@bernhard-42 Thank you very much! It works flawlessly ^^

Peque, 2019-12-18:
@bernhard-42 Trying out a simple assembly-to-HTML code:

```python
from cadquery import Workplane
from ipywidgets import embed
from jupyter_cadquery.cad_view import CadqueryView
from jupyter_cadquery.cadquery import Assembly
from jupyter_cadquery.cadquery import Part

# Create a simple assembly
box1 = Workplane('XY').box(10, 10, 10).translate((0, 0, 5))
a1 = Assembly([Part(box1)], "example 1")

# Generate HTML
a1.collect_shapes()
view = CadqueryView()
for shape in a1.collect_shapes():
    view.add_shape(shape["name"], shape["shape"], shape["color"])
renderer = view.render()
embed.embed_minimal_html('export.html', views=renderer, title='Renderer')
```

I end up with the camera initially looking at `(0, 0, 0)` instead of `camera_target`. Once I interact with it, the view is correctly positioned to look at `camera_target` (at the center of the part):

[image]

Do you know how could I fix that?

bernhard-42, 2019-12-21:
@Peque Sorry, no idea. I don't use embedded widgets. Maybe the embedding` javascript code does not call `update()` at the beginning but when you interact it will - just a wild guess (it will be hidden somewhere in https://unpkg.com/@jupyter-widgets/html-manager@^0.18.0/dist/embed-amd.js)

Peque, 2019-12-23:
@bernhard-42 Thanks. :blush:

Opened an [issue in `ipywidgets`](https://github.com/jupyter-widgets/ipywidgets/issues/2649), in case they can shed some light. :crossed_fingers: :innocent:

Peque, 2019-12-29:
I also [asked a question in Stack Overflow](https://stackoverflow.com/questions/59462182/embed-widgets-with-jupyter-cadquery-threejs-wrong-position-on-load), with a bounty, but it seems it is not getting much attention. :sweat_smile:

jasongrout, 2019-12-30:
> (it will be hidden somewhere in [https://unpkg.com/@jupyter-widgets/html-manager@^0.18.0/dist/embed-amd.js](https://unpkg.com/@jupyter-widgets/html-manager@%5E0.18.0/dist/embed-amd.js))

Relevant source is in https://github.com/jupyter-widgets/ipywidgets/tree/master/packages/html-manager/src

bernhard-42, 2020-01-02:
@Peque 
It did take some time, but then I remembered. I had a similar problem with jupyter_cadquery:

https://github.com/bernhard-42/jupyter-cadquery/blob/c0e5e4a41c044e6e8dac2be14e08a0ddcdddf571/jupyter_cadquery/cad_view.py#L493-L497

I am not sure, but I think it got introduced when I switched to `CombinedCamera`.

Not that I have a solution now, but a stronger indication that a call to `update` might help:

https://github.com/bernhard-42/jupyter-cadquery/blob/c0e5e4a41c044e6e8dac2be14e08a0ddcdddf571/jupyter_cadquery/cad_view.py#L300-L302

Do you have any chance of calling a javascript function?

Peque, 2020-01-02:
@bernhard-42 Thanks for the update.

> Do you have any chance of calling a javascript function?

:woman_shrugging: 

Do not know much about JS. But will ask someone in case they know the answer to that question. :joy: 

I also had a quick look at the code @jasongrout pointed to, but did not understand what was going on or where to look. :sweat_smile:

bernhard-42, 2020-01-03:
@Peque I replaced the `zoom` / `update` cascade by `self.controller.reset()` and published `0.9.2`.
For me the export now works as expected.
Could you please give it a try?
Thanks

Peque, 2020-01-03:
@bernhard-42 I am getting the same results with 0.9.3.

Not only locally, but also with Binder. The code I am used is [the code shared before](https://github.com/bernhard-42/jupyter-cadquery/issues/5#issuecomment-567202449). I shared instructions on to reproduce it with Binder in [my StackOverflow question](https://stackoverflow.com/questions/59462182).

Can you share the code you are using so that I can try it locally and see if it works for me too?

bernhard-42, 2020-01-03:
If I remove the `translate(0, 0, 5)` from your code 
[image]
it works for me.

Peque, 2020-01-03:
@bernhard-42 Yeah, but that's cheating! :stuck_out_tongue_winking_eye: 

If you remove the Z translation, then the center of the cube (where the controller is correctly pointing to) is `(0, 0, 0)`. The problem is that the camera is initially pointing to `(0, 0, 0)`, but should be pointing to the center of the cube.

The perspective is also lost (i.e.: the exported cube does not use an orthogonal view).

bernhard-42, 2020-01-03:
ok, fair enough, let's keep `translate` for further analysis ;-) 

On the other note, I looked at the orthogonal issue and copied the example from https://github.com/jupyter-widgets/pythreejs/blob/master/examples/Combined%20Camera.ipynb and slightly changed it:

```python
from pythreejs import *
from IPython.display import display

mesh1 = Mesh(SphereBufferGeometry(10, 8, 8), MeshLambertMaterial(color='red', opacity=0.5, transparent=True), position=[-20, 0, 0])
mesh2 = Mesh(BoxBufferGeometry(20, 16, 16), MeshLambertMaterial(color='green', opacity=0.5, transparent=True), position=[20, 0, 0])

view_width = 600
view_height = 400
camera = CombinedCamera(position=[0, 0, 60], width=view_width, height=view_height)

key_light = PointLight(position=[-100, 100, 100])
ambient_light = AmbientLight(intensity=0.4)
scene = Scene(children=[mesh1, mesh2, key_light, ambient_light, camera])
renderer = Renderer(scene=scene, camera=camera, controls=[OrbitControls(controlling=camera)],
                    width=view_width, height=view_height)
camera.mode = 'orthographic'
embed.embed_minimal_html('export4.html', views=renderer, title='Renderer')
display(renderer)
```

It produces an orthographic view in Jupyter but a perspective view in the export.html. Look's like CombinedCamera doesn't work for embedding.

Unfortunately, I don't get OrthographicCamera to work (it doesn't allow to rotate and I have issues with the parameters) ...

Peque, 2020-01-03:
@bernhard-42 That is great! Being able to get `jupyter-cadquery` out of the equation. :blush:

Would you mind copying that last message on https://github.com/jupyter-widgets/ipywidgets/issues/2649 ? I think it makes more sense there.

bernhard-42, 2020-01-03:
@Peque I adapted the pure pythreejs sample to be similar to your problem. After exporting, the view does not jump ...
Looks like this is something hidden in jupyter_cadquery :-(

```python
from pythreejs import *
from IPython.display import display

mesh = Mesh(BoxBufferGeometry(10, 10, 10), 
            MeshLambertMaterial(color='green', opacity=0.5, transparent=True), 
            position=[0, 5, 0])

view_width = 600
view_height = 400
camera = CombinedCamera(position=[0, 0, 60], width=view_width, height=view_height)

key_light = PointLight(position=[-100, 100, 100])
ambient_light = AmbientLight(intensity=0.4)
scene = Scene(children=[mesh, key_light, ambient_light, camera])
renderer = Renderer(scene=scene, camera=camera, controls=[OrbitControls(controlling=camera)],
                    width=view_width, height=view_height)
camera.mode = 'orthographic'
camera.zoom = 4
embed.embed_minimal_html('export9.html', views=renderer, title='Renderer')
display(renderer)
```

bernhard-42, 2020-01-04:
Thanks @Peque for reproducing the jumping issue without jupyter-cadquery!

bernhard-42, 2020-01-18:
@Peque I updated your issue in the ipywidgets repository with my findings when I rewrote the jupyter-cadquery viewing code.
I hope the issue is now clearer ...

bernhard-42, 2021-03-06:
Unfortunately, still not fixed in pythreejs :-(

bernhard-42, 2021-05-19:
@Peque 
This issue is quite old, isn't it? However, I recently published `orbitcontrol-patch`(see https://github.com/jupyter-widgets/pythreejs/issues/308#issuecomment-843828555) and integrated it into *Jupyter-CadQuery* - available in the latest release candidate of v2.2.0.

Exports now work as expected. 

You might not be interested any more after such a long time, however, I wanted to let you know that this did not get forgotten :-)

Peque, 2021-05-19:
@bernhard-42 Thanks for the update! Due to time constraints at this moment I will not be able to work on this, but it is good to know. :blush:

bernhard-42, 2022-03-13:
In the meantime with Jupyter CadQuery 3.0.0 the whole CAD view can be exported to a web page:

[image]

Hence, closing the issue

closed 2022-03-13 by bernhard-42