bernhard-42/jupyter-cadquery #6: massive multi body assembly

CPME, 2019-10-27:
I cannot view an entire assembly of 900 elements (see attached image). 400 element assembly is fully viewable. How many elements show up seems indeterminate, as it is different every time I run the script
[image]
. Interpreter does not throw error. Here is the code that generates chainmail.

`import cadquery as cq
from jupyter_cadquery.cadquery import (Assembly, Part, Edges, Faces, Vertices, show)
from jupyter_cadquery import set_sidecar

set_sidecar("CadQuery")

ring_od = 5
ring_id = 4
ring_t = 1
clearance = 1
ring_offset = ring_od+clearance
n_x = 30
n_y = 30
x_hat = (1, 0, 0)
y_hat = (0, 1, 0)
z_hat = (0, 0, 1)

ring = cq.Workplane('XY').circle(ring_od).circle(ring_id).extrude(ring_t)
ringx = ring.rotate((0, 0, 0), (1, 0, 0), 90).translate((ring_offset, 0, 0))
ringy = ring.rotate((0, 0, 0), (0, 1, 0), 90).translate((0, ring_offset, 0))
ringy2 = ring.rotate((0, 0, 0), (0, 1, 0), 90).translate((0, 2*ring_offset, 0))
unit_cell = ring.union(ringy).union(ringx)
#mail = unit_cell.translate((ring_offset, 0, 0))
cells_x = [unit_cell.translate((2 * x * ring_offset, 0, 0)) for x in range(n_x)]
cells_y = [[i.translate((0, 2 * y * ring_offset, 0)) for i in cells_x] for y in range(n_y)]
flattened_list = [y for x in cells_y for y in x]
#column = cq.CQ.add()
#ringy.add([ringx, ringy])
#print(ringy.all())
#ring = cq.CQ.add(ringx)
rings = [ring, ringx, unit_cell]

def assemble(parts=[]):
    return Assembly([Part(part, "red box",   "#d7191c", show_edges=False) for part in parts],"test")

ring_assembly = assemble(flattened_list)

show(ring_assembly, axes=True)
`

CPME, 2019-10-27:
Ran it again with no modifications, and it generated the entire model.
[image] Any ideas?

bernhard-42, 2019-10-27:
@CPME Firstly, thanks for giving jupyter-cadquerya try!

I would have expected that memory is an issue, so I tried your example on an (already swapping) 8GB mac laptop and a bit 128GB linux machine. In both cases the model rendered fine.

I then moved on to 50x50 rings on the linux machine and it stopped in the middle. 

[image]

However I received the error from jupyterlab

[image]

Did you see something like this? Would changing iopub rates on your machine help?

CPME, 2019-10-28:
@bernhard-42 Thank you for developing this nifty tool. It helped me get cad-query up and running pretty easily. I didn't receive an error at all. I'm on a 16 gb linux machine (fedora). i'm trying the 50X50 array now.

bernhard-42, 2020-08-16:
Closing, since no question left

closed 2020-08-16 by bernhard-42