bernhard-42/jupyter-cadquery #7: cqparts example fails

westurner, 2019-11-05:
With the docker image:

```python
import cqparts
```

Returns:

```
ModuleNotFoundError: No module named 'cqparts'
```

cqparts is not listed in environment.yml.
- ``!pip install cqparts`` fails because FreeCAD is not installed.
- https://github.com/cqparts/cqparts/issues/98#issuecomment-397875169 references a fork at https://github.com/CapableRobot/cadquery pending completion of the "OpenCascade (OCC)" milestone: https://github.com/cqparts/cqparts/milestone/2
  - ``!pip install --user -e git+https://github.com/CapableRobot/cqparts#egg=cqparts`` fails because there's no setup.py in the cqparts repo?
    - "streamline release process" https://github.com/cqparts/cqparts/issues/79
- ``!cd /opt/conda/envs/cq/lib/python3.6/site-packages; git clone https://github.com/CapableRobot/cqparts`` fails because we don't have write permissions to the cq env's site-packages

Adding this cell to the top of cqparts.ipynb seems to work:

```
!rm -rf cqparts*
!test -d _cqparts_src || git clone https://github.com/CapableRobot/cqparts ./_cqparts_src
!ln -s _cqparts_src/src/* .
import cqparts
```

bernhard-42, 2019-11-07:
Hi @westurner 
you are describing exactly my journey with cqparts and I gave up with it.
I don't know whether @fragmuffin will continue with the project or whether it is finally on hold. So didn't want to spend more time on it until there is an officially published version ...

westurner, 2019-11-09:
Yeah, IDK what their timetable is on the OCC milestone.

Note that I've updated the provisional install of the CapableRobot fork to install into site.USER_SITE instead of the current directory:

```
USER_SITE = __import__("site").USER_SITE
!rm -rf $USER_SITE/cqparts*
!test -d _cqparts_src || git clone https://github.com/CapableRobot/cqparts ./_cqparts_src
!test -d $USER_SITE || mkdir -p $USER_SITE
!ln -s _cqparts_src/src/* $USER_SITE/
# !pip install -r _cqparts_src/src/cqparts/requirements.txt
!pip install tinydb  # everything else is already installed by conda
import cqparts 
```

bernhard-42, 2021-03-06:
@westurner cqparts seems pretty unmaintained. Not sure what to do with ticket :-)
btw. just release 2.0.0 with CadQuery 2.1 and JL 3.0

westurner, 2021-03-06:
From the updated README: 

> Assemblies
> - Supports CadQuery Assemblies
> - Support Manual Assemblies with animation of models

👍

closed 2021-03-06 by westurner