bernhard-42/jupyter-cadquery #9: Edges display

Peque, 2020-01-15:
It seems that convex edges are displayed/highlighted with a gray line, but concave edges are not.

In example, with this code the edges between the two extrusions are not displayed:

```python
from cadquery import Workplane

Workplane("XY").rect(2, 2).extrude(1).faces(">Z").rect(1, 1).extrude(1)
```

[image]

Is that the expected behavior? How could it be fixed?

bernhard-42, 2020-01-18:
@Peque Hmm, that is not nice indeed. I need to check whether I can control it in either cadquery, occ or pythreejs ....

bernhard-42, 2020-01-18:
@Peque Luckily this can be solved, it's z-fighting and threejs has a config for that. You can test it like that:

```python
tower = cq.Workplane("XY").rect(2, 2).extrude(1).faces(">Z").rect(1, 1).extrude(1)

r = show(tower)

b = r.cq_view.pickable_objects.children[0]
b.material.polygonOffset = True
b.material.polygonOffsetUnits = 0.5
b.material.polygonOffsetFactor = 1
```

[image]

I will add it into the next release

bernhard-42, 2020-01-18:
@Peque Version 0.9.4 should include this fix.
Additionally I rewrote the viewing routine, since it did not work properly when the center of the bounding box was not (0,0,0). This also looks better now.

Peque, 2020-01-18:
@bernhard-42 Thanks! I will try it today and report back. :blush:

Peque, 2020-01-18:
Some feedback. :innocent: 

Edges
--------

The edges seem to be fixed for that simple example now. Thanks! :blush: 

It does not seem to work for different parts that intersect though:

```python
from cadquery import Workplane
from jupyter_cadquery.cadquery import Assembly
from jupyter_cadquery.cadquery import Part

tower1 = Workplane("XY").rect(2, 2).extrude(1)
tower2 = Workplane("XY").rect(1, 1).extrude(2)

Assembly([Part(tower1), Part(tower2)])
```

[image]

Or is that expected?

Grid and axis
-----

I do not like the fact that the origin of the grid helper and axis helper is not at `(0, 0, 0)` now. In this example:

```python
Workplane("XY").rect(2, 2).extrude(1).faces(">Z").rect(1, 1).extrude(1)
```

I would expect the grid and the axis helpers to be bellow the tower, but they are crossing it:

[image]

bernhard-42, 2020-01-18:
For the first issue: 
There are actually no edges from a model perspective. If you switch on transparency 
[image]

or rotate to the bottom:

[image]

you can see the actual edges of the model. I don not modify the model in my previewer, I only show what I get. If you union the result `tower1.union(tower2)`, it is fine again:

[image]

bernhard-42, 2020-01-18:
For the second issue:
This is just the default setting where the axis are centered at the bounding box center of the object. This is btw. also the point where the model rotates around. 
To adapt, try:

```python
from jupyter_cadquery.cadquery import show
show(Workplane("XY").rect(2, 2).extrude(1).faces(">Z").rect(1, 1).extrude(1), 
     axes0=True, grid=True, axes=True, transparent=True)
```

which shows:

[image]

Is this what you want?

Peque, 2020-01-18:
Yeah, that solves all my issues. :tada:

I would still prefer the grid at `(0, 0, 0)` as the default behavior (feels more natural), but since I guess it is hard to do a survey and it is your project, you decide. :smile: 

Thanks a lot! :blush:

closed 2020-01-18 by Peque

bernhard-42, 2020-01-18:
How about:
```python
from jupyter_cadquery.cadquery import show as show_orig
def show(*shapes):
    show_orig(*shapes, axes0=True, grid=True, axes=True, transparent=True)
```
?

Peque, 2020-01-18:
@bernhard-42 Yeah, it is more like a philosophical preference. :joy:  Not that is hard for me to patch it. But thanks! :heart:

bernhard-42, 2020-01-18:
Got it. 
To be honest, I also don't know what I like more. I think I changed it because I played with some models that were far away from the origin and then I did not see grid or axis without zooming.

Let me play with it some more to find out my own preference  ...

bernhard-42, 2020-02-09:
Hi @Peque 
I realised that I would love to have different defaults based on what I currently do. So I have added a kind of defaults management to jupyter-cadquery. You now can globally set your preferences:

```python
import cadquery as cq
from jupyter_cadquery.cadquery import Part, Assembly, show
from jupyter_cadquery import set_defaults, get_defaults, reset_defaults

set_defaults(axes=True, grid=True, axes0=True, transparent=True, cad_width=1000, position=(1,1,0))
```

Every following rendering will use these defaults. And you also can use `show(cad_objs, grid=False)` to overwrite the default for one specific plot.

It is available in version `0.9.5` and I hope that addresses your "philosophical preference" :wink: