bernhard-42/jupyter-cadquery #10: Docker instructions incomplete/don't work

goatchurchprime, 2020-02-14:
I don't get along well with this newfangled docker thing yet, so I don't know my way round it other than typing in exactly the commands I'm told.  

It seems I need to check out the repo first to get the Dockerfile.  

> git clone git@github.com:bernhard-42/jupyter-cadquery.git
> cd jupyter-cadquery

And then it get stuck at a stage where files in /var/lib/ don't exist.  Am I supposed to run this sudo or something?

> cd docker
> docker build -t bernard-42/jupyter-cadquery:0.9.5 .

Sending build context to Docker daemon   2.56kB
Step 1/13 : FROM continuumio/miniconda3:4.7.10
 ---> 4a51de2367be
Step 2/13 : RUN adduser --disabled-password --gecos "Default user" --uid 1000 cq &&     apt-get update -y &&     apt-get install --no-install-recommends -y libgl1-mesa-glx libglu1-mesa &&     rm -rf /var/lib/apt/lists/* &&     mkdir /data && chown cq:cq /data
 ---> Using cache
 ---> e5233e26bfb4
Step 3/13 : ADD --chown=cq:cq examples /home/cq
ADD failed: stat /var/lib/docker/tmp/docker-builder775350372/examples: no such file or directory

bernhard-42, 2020-02-14:
I moved the Dockerfile into its own folder to separate it, without thinking about the dependencies.

Try the following:

```
cd jupyter-cadquery
ln -s docker/Dockerfile Dockerfile  # link it to the top level
docker build -t bernard-42/jupyter-cadquery:0.9.5 .
rm Dockerfile  # delete link
```

With this approach I was able to build the docker image.

I will fix it in the next release ...

goatchurchprime, 2020-02-17:
That got a bit further:

```
Step 13/13 : CMD bash -l -c "source /opt/conda/bin/activate cq && jupyter lab --ip=0.0.0.0 --no-browser --NotebookApp.token='' --NotebookApp.allow_origin='*'"
 ---> Using cache
 ---> fbe13d410e8f
Successfully built fbe13d410e8f
Successfully tagged bernard-42/jupyter-cadquery:0.9.5
```

But now there's a problem with the next command:

> docker run -it --rm -v /tmp/jupyter:/data/workdir -p 8888:8888 bernhard-42/jupyter-cadquery:0.9.5

```Unable to find image 'bernhard-42/jupyter-cadquery:0.9.5' locally
docker: Error response from daemon: pull access denied for bernhard-42/jupyter-cadquery, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
```

bernhard-42, 2020-02-17:
I see
„Successfully tagged bernard-42/jupyter-cadquery:0.9.5“
where bernard is without „h“ and in the run command bernhard with „h“

Could there be a typo?

bernhard-42, 2020-08-16:
@goatchurchprime I have updated my Jupyter Cadquery extension https://github.com/bernhard-42/jupyter-cadquery/ to v1.0.0:
It now supports CadQuery 2.0 and JupyterLab 2.2.4 
It is available from docker hub (image bwalter42/jupyter_cadquery:1.0.0)

nedtwigg, 2020-10-20:
FWIW, I am using the published docker image with great success.  Thanks very much for publishing, it's a nice, repeatable workflow (I am also a docker noob).

As your README states, just:

```
docker run -it --rm -v PATH_TO_DIRECTORY_I_WANT_TO_WORK_ON:/home/cq -p 8888:8888 bwalter42/jupyter_cadquery:1.0.0
```

bernhard-42, 2020-10-20:
@nedtwigg glad it works for you!

bernhard-42, 2020-11-22:
I guess this can be closed, please reopen if you have issues again

closed 2020-11-22 by bernhard-42