bernhard-42/jupyter-cadquery #13: Nothing displayed when a shape is 800mm wide

goatchurchprime, 2020-04-30:
If I run the following, I get a short prism drawn, and it works fine.
 
```Python
import cadquery as cq
from jupyter_cadquery.cadquery import show

r = cq.Workplane("XZ", (0,0,0))
r = r.moveTo(0,0)
r = r.lineTo(400,0)  # <--- change to 800 and there is nothing displayed!!!
r = r.lineTo(10,300)
r.close()
r = r.extrude(13)
show(r, axes=True, grid=True, ortho=True, axes0=True)
```

But if I change the 400 to 800, then nothing is drawn, not even any axes.

I've checked that the same problem occurs in that jupyter-cadquery launch|binder thing linked in the readme.

The exportSTL works when applied to the part, so it is there, but I don't know why the display fails.

bernhard-42, 2020-08-16:
@goatchurchprime Sorry, I completely missed this issue. Yes it is a bug. If you use 600, you can see parts of the triangle, the is clipped away. I need to check the camera settings

bernhard-42, 2020-08-16:
A quick workaround is:

```python
c = show(r, axes=True, grid=True, ortho=True, axes0=True)
c.cq_view.camera.far = 4000 # default is 2000
c.cq_view.camera.orthoFar = 4000 # default is 2000
```
with the far parameter depending on size of the object.

I need to change the default based on the bounding box

bernhard-42, 2021-03-06:
Fixed with 2.0.0.
Will close now, if you find further issues with this, please reopen

closed 2021-03-06 by bernhard-42