bernhard-42/jupyter-cadquery #14: Small example crash in jupyter_cadquery.cadquery.show()

goatchurchprime, 2020-04-30:
This tiny example probably puts `jupyter_cadquery.cadquery.show()` into an infinite loop.  

```Python
import cadquery as cq

r = cq.Workplane("XZ", (0,0,0))
r.moveTo(0, 0).lineTo(0.570634, 0.082918).lineTo(0.352671, 0.217082).lineTo(-0.352671, 0.217082).close()
r = r.workplane(offset=0.2, centerOption="ProjectedOrigin")
r.moveTo(0, 0).lineTo(0.475528, 0.069098).lineTo(0.293893, 0.180902).lineTo(-0.293893, 0.180902).close()
r = r.loft(combine=True)

from jupyter_cadquery.cadquery import show
show(r)   # <--- this now hangs at 100% on all four cores
```

To be fair the script is invalid because two of the lines should start with `r = r.moveTo(...)`, but I hit this case with my buggy code, and found a crash that shouldn't happen.

bernhard-42, 2020-08-16:
@goatchurchprime In your first example the small numbers kill the tesselation algorithm I use from python-occ (even with CadQuery 2.0, but I haven't tested with the OCP branch).
What works is to multiply the values by 10

```
r = cq.Workplane("XZ", (0,0,0))
r = r.moveTo(0, 0).lineTo(5.70634, 0.82918).lineTo(3.52671, 2.17082).lineTo(-3.52671, 2.17082).close()
r = r.workplane(offset=2, centerOption="ProjectedOrigin")
r = r.moveTo(0, 0).lineTo(4.75528, 0.69098).lineTo(2.93893, 1.80902).lineTo(-2.93893, 1.80902).close()
r = r.loft(combine=True)

show(r)
```
It renders on my machine in 500ms. I have seen this before, and don't have an answer atm how to tesselate small scale objects.

For the second issue, 
```python
from jupyter_cadquery import set_defaults
set_defaults(edge_accuracy=0.02, quality=0.02
``` 
creates the rendering in 600ms

bernhard-42, 2020-11-22:
will be tracked with #19

closed 2020-11-22 by bernhard-42