bernhard-42/jupyter-cadquery #15: Docker container built as per examples fails to run example notebook

mitchins, 2020-07-09:
I checked out the container and build it (Ensuring all files are copied from current workdir etc)
It can build and deploy but when running the cadquery_example.ipynb

```
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-dce0531510f8> in <module>
----> 1 import cadquery as cq
      2 from jupyter_cadquery.cadquery import (Assembly, Part, Edges, Faces, Vertices, show, 
      3                                        replay, enable_replay, disable_replay, reset_replay)
      4 from jupyter_cadquery import set_sidecar
      5 

/opt/conda/envs/cq/lib/python3.7/site-packages/cadquery/__init__.py in <module>
      1 # these items point to the OCC implementation
----> 2 from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
      3 from .occ_impl.shapes import (
      4     Shape,
      5     Vertex,

/opt/conda/envs/cq/lib/python3.7/site-packages/cadquery/occ_impl/geom.py in <module>
      3 from typing import overload, Sequence, Union, Tuple, Type, Optional
      4 
----> 5 from OCP.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp_GTrsf, gp, gp_XYZ
      6 from OCP.Bnd import Bnd_Box
      7 from OCP.BRepBndLib import BRepBndLib

ModuleNotFoundError: No module named 'OCP'
```

Just wondering if there's something that's changed?

I've actually had alot of issues trying to install this myself, and thought I'd try the Docker container for a bullet proof reference, but out of the box it's got at least this issue.

Did I miss something, or is there a caveat here?
As there aren't any prebuilt images it's possible some package or platform has changed? But they all appear to be locked in the environment.yml

bernhard-42, 2020-07-12:
I guess it is this line in `environment.yml`: `   - git+https://github.com/CadQuery/cadquery.git`
I don't have time at the moment to test or change, but after master switching to `OCP` this is broken.
Maybe try `   - cadquery` to use cadquery from conda channel `cadquery` which is based on `pyhtonocc`

bernhard-42, 2020-08-16:
@mitchins I have updated my Jupyter Cadquery extension https://github.com/bernhard-42/jupyter-cadquery/ to v1.0.0:
It now supports CadQuery 2.0 and JupyterLab 2.2.4 and it is available from docker hub (image bwalter42/jupyter_cadquery:1.0.0, so for normal use cases no need to create your own container any more.

nedtwigg, 2020-10-20:
I am also having trouble with examples that use CadQuery's built-in assemblies.  It seems like Jupyter-CadQuery has its own independent assembly mechanism, and the [cadquery-assemblies](https://github.com/bernhard-42/jupyter-cadquery/blob/master/examples/cadquery-assemblies.ipynb) notebook has some utilities to translate between them.

However, anytime I call `cq.Assembly`, I get `AttributeError: module 'cadquery' has no attribute 'Assembly'`.  It looks like quite some work has taken place [since `1.0.0` was published](https://github.com/bernhard-42/jupyter-cadquery/compare/v1.0.0...master), I wonder if the docker image has a too-old version of CadQuery.

- any chance of a `v1.1.0` release with the latest changes?
- any chance that the `convert / show_assy` functions will get subsumed into core, so I can just `show(any cq object)`
- this project is super cool.  I'm a hobbyist tinkering with it to [maybe a build a shed](https://github.com/nedtwigg/nedshed)

bernhard-42, 2020-10-20:
@nedtwigg The problem is how to follow the development of `cadquery`. Currently I support the latest published version of `cadquery` with `jupyter-cadquery`. The latest master of `cadquery` is based on `OCP` instead of `pythonocc` and it took some time to port my stuff to it. Assemblies in `jupyter-cadquery` were invented as simple groups of objects without linking mechanism. Later then `cadquery` came up with real assemblies. As soon as `cadquery` releases the next version including all this (and I don't know the plans of @adam-urbanczyk), `jupyter-cadquery` will follow (including full `cq.Assembly` support. 

Currently the only way to use `cadquery` master with `jupyter-cadquery` master is to follow https://github.com/bernhard-42/jupyter-cadquery#development-setup and use `convert/show_assy`.

nedtwigg, 2020-10-20:
Understood.  When I run your docker image, I don't need to install anything at all on my machine.  Totally reasonable for your publishing policy to be "we publish images when our dependencies publish stable versions".  But if you wanted to, you could publish `1.1.0.BETA` based on the cadquery master branch, and it would be perfectly repeatable from a "consume-the-image" standpoint, although granted it would not be repeatable from a "produce-the-image" standpoint, but docker images rarely are.

From my perch as a novice user, it doesn't matter very much / at all if the versions underneath the image are stable (since the image freezes their contents regardless), but it is helpful if the examples work.  Also as a novice user, pip and conda have eaten a lot of my time with unsatisfying result, but your docker image is tremendously productive.  But I'm very happy with the `1.0` image you have given us already, so I'm a happy camper regardless :)

bernhard-42, 2020-10-20:
yeah, could be an option to just release a "follow-master" docker image additionally to the stable docker image whenever something significantly changes in `cadquery` or `jupyter-cadquery`. 
I've opened a separate issue for this, https://github.com/bernhard-42/jupyter-cadquery/issues/28

adam-urbanczyk, 2020-10-20:
I'd expect a release by the end of this year.

Regarding conda issues feel free to ask for help, but in general: install miniconda to a local dir, do not perform conda init and initialize via `[scripts,bin]/activate`.

bernhard-42, 2021-03-06:
@nedtwigg With the release of 2.0.0, can we close this now?

nedtwigg, 2021-03-06:
I just pulled master and tried to run the examples, huge improvement thanks very much!! A few issues I bumped into:

- `examples/assemblies/1-disk-arm.ipynb`: ModuleNotFoundError: No module named 'matplotlib'
- `examples/assemblies/3-jansen-linkage.ipynb`: ModuleNotFoundError: No module named 'matplotlib'
- `examples/assemblies/5-door.ipynb`: requires downloading a DXF file. Not a big deal, but might be nice if this were included in the repo (only 58KB) so that the examples keep working after the URL inevitably goes stale.

It's easy enough to ignore the matplotlib cells, but might be nice include in future releases. Fine with me to close this issue.

bernhard-42, 2021-03-06:
Thanks for the feedback. I guess for the docker container I should add matplotlib, right. At the moment, just add a cell with 
```python
%pip install matplotlib
```
run it and then restart the kernel. Afterwards the matplotlib cells are good to go.

For the dxf files, I am not sure what the license is so I did not want to add it to the container to not violate any copyright.

bernhard-42, 2022-03-13:
solved in the meantime, e.g. `docker run -it --rm -v $WORKDIR:/home/cq -p 8888:8888 bwalter42/jupyter_cadquery:3.0.0`

closed 2022-03-13 by bernhard-42