bernhard-42/jupyter-cadquery #18: Module Not Found Error

koojoobee, 2020-08-09:
**I tried importing CadQuery in my jupyterlab session and here is the error I get,** 

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-127b0e866cb5> in <module>
----> 1 import cadquery as cq

~/opt/anaconda3/envs/research3/lib/python3.7/site-packages/cadquery/__init__.py in <module>
      1 # these items point to the OCC implementation
----> 2 from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
      3 from .occ_impl.shapes import (
      4     Shape,
      5     Vertex,

~/opt/anaconda3/envs/research3/lib/python3.7/site-packages/cadquery/occ_impl/geom.py in <module>
      3 from typing import overload, Sequence, Union, Tuple, Type, Optional
      4 
----> 5 from OCP.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp_GTrsf, gp, gp_XYZ
      6 from OCP.Bnd import Bnd_Box
      7 from OCP.BRepBndLib import BRepBndLib

ModuleNotFoundError: No module named 'OCP'

---------------------------------------------------------------------------

**Do you have any solutions for this?**

bernhard-42, 2020-08-16:
@koojoobee I have updated the Jupyter Cadquery extension to v1.0.0: It now supports CadQuery 2.0 and JupyterLab 2.2.4

bernhard-42, 2020-11-22:
I guess this can be closed, please reopen if you still have issues

closed 2020-11-22 by bernhard-42