bernhard-42/jupyter-cadquery #20: Permission Denied Error When Running New Docker Image

jmwright, 2020-08-17:
I get a permission denied error when running the 1.0.0 image published yesterday.

System info:
Ubuntu 18.04
Docker version 19.03.12, build 48a66213fe

I pulled the 1.0.0 image via this command:
```
docker pull bwalter42/jupyter_cadquery:1.0.0
```
I then tried to run the following from the readme.
```
WORKDIR=/tmp/jupyter
docker run -it --rm -v $WORKDIR:/home/cq -p 8888:8888 bwalter42/jupyter_cadquery:1.0.0
```
And got this error:
```
PermissionError: [Errno 13] Permission denied: '/home/cq/.local'
```
Here's the full stack trace.
```
Traceback (most recent call last):
  File "/opt/conda/envs/cq/lib/python3.7/site-packages/traitlets/traitlets.py", line 528, in get
    value = obj._trait_values[self.name]
KeyError: 'runtime_dir'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/conda/envs/cq/bin/jupyter-lab", line 10, in <module>
    sys.exit(main())
  File "/opt/conda/envs/cq/lib/python3.7/site-packages/jupyter_core/application.py", line 270, in launch_instance
    return super(JupyterApp, cls).launch_instance(argv=argv, **kwargs)
  File "/opt/conda/envs/cq/lib/python3.7/site-packages/traitlets/config/application.py", line 663, in launch_instance
    app.initialize(argv)
  File "<decorator-gen-7>", line 2, in initialize
  File "/opt/conda/envs/cq/lib/python3.7/site-packages/traitlets/config/application.py", line 87, in catch_config_error
    return method(app, *args, **kwargs)
  File "/opt/conda/envs/cq/lib/python3.7/site-packages/notebook/notebookapp.py", line 2026, in initialize
    self.init_configurables()
  File "/opt/conda/envs/cq/lib/python3.7/site-packages/notebook/notebookapp.py", line 1555, in init_configurables
    connection_dir=self.runtime_dir,
  File "/opt/conda/envs/cq/lib/python3.7/site-packages/traitlets/traitlets.py", line 556, in __get__
    return self.get(obj, cls)
  File "/opt/conda/envs/cq/lib/python3.7/site-packages/traitlets/traitlets.py", line 535, in get
    value = self._validate(obj, dynamic_default())
  File "/opt/conda/envs/cq/lib/python3.7/site-packages/jupyter_core/application.py", line 100, in _runtime_dir_default
    ensure_dir_exists(rd, mode=0o700)
  File "/opt/conda/envs/cq/lib/python3.7/site-packages/jupyter_core/utils/__init__.py", line 13, in ensure_dir_exists
    os.makedirs(path, mode=mode)
  File "/opt/conda/envs/cq/lib/python3.7/os.py", line 213, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "/opt/conda/envs/cq/lib/python3.7/os.py", line 213, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "/opt/conda/envs/cq/lib/python3.7/os.py", line 213, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "/opt/conda/envs/cq/lib/python3.7/os.py", line 223, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/home/cq/.local'
```

bernhard-42, 2020-08-17:
Under Linux it looks like docker would create `/tmp/jupyter` if it does not exist. And it creates it with root:root as owner. I wasn't aware of this.

```bash
$ ls -ld /tmp/jupyter/
drwxr-xr-x 2 root root 4096 Aug 17 21:55 /tmp/jupyter/
```

Two choices:

1. `sudo chown $USER /tmp/jupyter/``
2. delete `/tmp/jupyter` and create it manually before starting the docker command

I need to update the readme to tell that WORKDIR needs to exist

bernhard-42, 2020-08-17:
updated the README

jmwright, 2020-08-17:
Works great for me. Thanks for the quick response.

closed 2020-08-17 by jmwright