bernhard-42/jupyter-cadquery #25: No model rendered

ahupp, 2020-10-13:
I setup a fresh environment using the instructions in the README, and am trying to render the example there "b) Example: CadQuery using Sidecar".  This gives me the sidecar (a new panel on the left-hand side that labeled "CadQuery"), a "Tree" panel and a "Rendering Done" panel.  However, no rendering appears.   This is in Chrome 86 on Windows, running against jupyter-cadquery on linux.

Sorry for not including versions, but I'm not familiar enough with this distribution system to know what to include.  

There are some threejs errors on the console:

  | getInverse | @ | three.module.js:1532
-- | -- | -- | --
  | updateMatrixWorld | @ | three.module.js:20991
  | updateMatrixWorld | @ | three.module.js:8747
  | WebGLRenderer.render | @ | three.module.js:22946
  | renderScene | @ | Renderable.js:383
  | tock | @ | Renderable.js:268
  | requestAnimationFrame (async) |   |  
  | tick | @ | Renderable.js:261
  | triggerEvents | @ | backbone.js:370
  | triggerApi | @ | backbone.js:356
  | eventsApi | @ | backbone.js:155
  | Events.trigger | @ | backbone.js:346
  | onChildChanged | @ | Renderer.js:59
  | triggerEvents | @ | backbone.js:369
  | triggerApi | @ | backbone.js:356
  | eventsApi | @ | backbone.js:155
  | Events.trigger | @ | backbone.js:346
  | onChildChanged | @ | Three.js:359
  | triggerEvents | @ | backbone.js:369
  | triggerApi | @ | backbone.js:356
  | eventsApi | @ | backbone.js:155
  | Events.trigger | @ | backbone.js:346
  | onChildChanged | @ | Three.js:359
  | triggerEvents | @ | backbone.js:370
  | triggerApi | @ | backbone.js:356
  | eventsApi | @ | backbone.js:155
  | Events.trigger | @ | backbone.js:346
  | set | @ | backbone-patch.js:118
  | push.rynU.WidgetModel.set | @ | widget.js:295
  | push.rynU.WidgetModel.set_state | @ | widget.js:229
  | (anonymous) | @ | widget.js:213
  | Promise.then (async) |   |  
  | push.rynU.WidgetModel._handle_comm_msg | @ | widget.js:212
  | _handleCommMsg | @ | default.js:1028
  | _handleMessage | @ | default.js:1144
  | async function (async) |   |  
  | _handleMessage | @ | default.js:1096
  | (anonymous) | @ | default.js:90
  | Promise.then (async) |   |  
  | KernelConnection._onWSMessage | @ | default.js:87

And then this when I try to interact with the empty space that presumably holds the render:

  | flatten | @ | three.module.js:15762
-- | -- | -- | --
  | setValueV4a | @ | three.module.js:16205
  | push.Womt.WebGLUniforms.upload | @ | three.module.js:16511
  | setProgram | @ | three.module.js:23822
  | WebGLRenderer.renderBufferDirect | @ | three.module.js:22555
  | renderObject | @ | three.module.js:23308
  | renderObjects | @ | three.module.js:23278
  | WebGLRenderer.render | @ | three.module.js:23026
  | renderScene | @ | Renderable.js:383
  | tock | @ | Renderable.js:268
  | requestAnimationFrame (async) |   |  
  | tick | @ | Renderable.js:261
  | triggerEvents | @ | backbone.js:370
  | triggerApi | @ | backbone.js:356
  | eventsApi | @ | backbone.js:155
  | Events.trigger | @ | backbone.js:346
  | onChildChanged | @ | Renderer.js:59
  | triggerEvents | @ | backbone.js:369
  | triggerApi | @ | backbone.js:356
  | eventsApi | @ | backbone.js:155
  | Events.trigger | @ | backbone.js:346
  | onChildChanged | @ | Three.js:359
  | triggerEvents | @ | backbone.js:370
  | triggerApi | @ | backbone.js:356
  | eventsApi | @ | backbone.js:155
  | Events.trigger | @ | backbone.js:346
  | set | @ | backbone-patch.js:118
  | push.rynU.WidgetModel.set | @ | widget.js:295
  | update_controlled | @ | OrbitControls.js:32
  | (anonymous) | @ | OrbitControls.js:21
  | dispatchEvent | @ | three.module.js:172
  | onMouseUp | @ | OrbitControls.js:804

bernhard-42, 2020-10-14:
My first guess is that pythreejs might not be installed properly. Could you test e.g.

```pyhton
from pythreejs import *
DodecahedronGeometry(radius=10, detail=0, _flat=True)
``` 

It should result in 

[image]

bernhard-42, 2020-10-14:
In my conda environment for jupyter-cadquery, I see:

```bash
$ jupyter-labextension list
JupyterLab v2.2.6
Known labextensions:
   app dir: /opt/miniconda/envs/cq2-jl/share/jupyter/lab
        @jupyter-widgets/jupyterlab-manager v2.0.0  enabled  OK
        @jupyter-widgets/jupyterlab-sidecar v0.5.0  enabled  OK
        jupyter-threejs v2.2.0  enabled  OK
        jupyter_cadquery v1.0.0  enabled  OK
        jupyterlab-datawidgets v6.3.0  enabled  OK
```

and

```bash
$ conda list | egrep '(pythreejs|jupyter|IPython|ipywidgets)'
ipywidgets                7.5.1                      py_1
jupyter-cadquery          1.0.0                    pypi_0    pypi
jupyter_client            6.1.7                      py_0
jupyter_core              4.6.3                    py37_0
jupyterlab                2.2.6                      py_0
jupyterlab_pygments       0.1.2                      py_0
jupyterlab_server         1.2.0                      py_0
pythreejs                 2.2.1              pyh9f0ad1d_0    conda-forge
```

ahupp, 2020-10-14:
This is what I have:

```
 %  jupyter-labextension list
JupyterLab v2.2.6
Known labextensions:
   app dir: /home/ahupp/.conda/envs/cadquery-jl/share/jupyter/lab
        @jupyter-widgets/jupyterlab-manager v2.0.0  enabled  OK
        @jupyter-widgets/jupyterlab-sidecar v0.5.0  enabled  OK
        jupyter-threejs v2.2.0  enabled  OK
        jupyter_cadquery v1.0.0  enabled  OK
        jupyterlab-datawidgets v6.3.0  enabled  OK
ahupp@LAPTOP-KQ5N2P8K ~/cadexp
 %  conda list | egrep '(pythreejs|jupyter|IPython|ipywidgets)'
ipywidgets                7.5.1                      py_1
jupyter-cadquery          1.0.0                    pypi_0    pypi
jupyter_client            6.1.7                      py_0
jupyter_core              4.6.3                    py37_0
jupyterlab                2.2.6                      py_0
jupyterlab_pygments       0.1.2                      py_0
jupyterlab_server         1.2.0                      py_0
pythreejs                 2.2.0              pyh8c360ce_0    conda-forge
```

bernhard-42, 2020-10-14:
That is pretty much the same. Did the pythreejs test work for you?

ahupp, 2020-10-14:
The pythreejs test did work.  I'm also seeing these errors (not sure why i didn't get them last time):

[image]

bernhard-42, 2020-10-14:
The "widget model not found" happens for example, when you refresh the page. It takes some time until model and javascript view are in synch (at least that's how I understand it)
But the other errors should definitely not appear.
I still think something internally went wrong during labextension build, but I cannot reproduce at the moment. 
In the past when I had these unclear error messages around pythreeejs in the javascript console, I uninstalled the labextensions, and installed them again starting from @jupyter-widgets/jupyterlab-manager. 
Sorry, not very convenient, but I have no better idea

ahupp, 2020-10-14:
I wiped out this conda environment and reinstalled everything from scratch, and it works now.   No idea what was wrong, but thanks for looking at it.

closed 2020-10-14 by ahupp

bernhard-42, 2020-10-14:
glad to hear it works for you now