bernhard-42/jupyter-cadquery #26: Inline result cells

nedtwigg, 2020-10-19:
The sidecar is very useful for tinkering.  But once things are working well, I would prefer to have inline result cells, so I could tell a full story in the notebook, with renderings of specific details as I go.  I think that #5 does a lot of the hard parts of this...

bernhard-42, 2020-10-19:
If you omit the `set_sidecar` after restarting the kernel, then sidecar will not be used.

Alternatively, run

```python
import jupyter_cadquery.cad_display as cd
cd.SIDECAR.close()
cd.SIDECAR = None
```

If you want to get back to sidecar, use `set_sidecar` again

nedtwigg, 2020-10-20:
Awesome, thanks very much!  These inline result cells work great in the notebook editor, but they are not visible in an HTML export.  If I understand correctly, #5 is about decoupling the 3D viewer from its notebook connection, and could possibly be reused for exporting a notebook to HTML including models?

bernhard-42, 2020-10-20:
@nedtwigg Understood, I think I should add something like that in the next release

bernhard-42, 2022-08-02:
Haven't really found a way to do that.
Two workarounds:
- export html and use in own web page
- use a cell CAD viewer and pin important view to output as PNG.

Closing as wontfix

closed 2022-08-02 by bernhard-42