bernhard-42/jupyter-cadquery #28: Create a docker image for the dev version

bernhard-42, 2020-10-20:
To avoid `conda/...` whenever there are significant changes in `cadquery` or `jupyter-cadquery`, release a docker image "-master"

nedtwigg, 2020-10-20:
If this happens, big vote for `1.1.0-master`, rather than moving `master` around.  Very helpful to have a repeatable setup.

bernhard-42, 2020-11-25:
@nedtwigg Good news, a dev version is available on docker hub `bwalter42/jupyter_cadquery:2.0.0-beta2` 
It comes with the latest CadQuery (including assemblies), supports assemblies out of the box and comes wth its own mate based assembly (`MAssemby`) which can be used to also animate your CAD Assemblies. (see gif on top of the Readme)
It will be stable, since new every docker images will get the version number of the git tag.
Hope that helps

nedtwigg, 2020-11-25:
Huzzah, thanks very much! I stopped work on my little project because of this hangup, and now I'll start it again!

bernhard-42, 2021-03-06:
@nedtwigg Just release 2.0.0, also as a docker image. It uses the latest stable CadQuery 2.1 now

bernhard-42, 2021-05-15:
Closing, as I regularly release new docker containers now

closed 2021-05-15 by bernhard-42