bernhard-42/jupyter-cadquery #29: Error when trying to export

sergi, 2020-11-20:
I have the following cadquery example code:

```python
import cadquery as cq
from jupyter_cadquery.cadquery import Assembly, Part, Edges, Faces, Vertices, show
from jupyter_cadquery import set_sidecar, set_defaults, reset_defaults

set_sidecar("CadQuery")  # force usage of one cad view on the right
set_defaults(axes=False, grid=True, axes0=True, ortho=True, transparent=True) # Set default values

height = 60.0
width = 80.0
thickness = 10.0
diameter = 22.0
padding = 12.0

 # make the base
result = cq.Workplane("XY").box(height, width, thickness)\
 .faces(">Z").workplane().hole(diameter)\
 .faces(">Z").workplane() \
 .rect(height - padding, width - padding, forConstruction=True)\
 .vertices().cboreHole(2.4, 4.4, 2.1)\
 .edges("|Z").fillet(2.0)

show(result)

cq.exporters.export(result,'result.stl')
```

It all works great for visualizing, but when I try to export (last line) I get the following error:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-9-09152fc940df> in <module>
     23 show(result)
     24 
---> 25 cq.exporters.export(result,'result.stl')

AttributeError: module 'cadquery.occ_impl.exporters' has no attribute 'export'
```

As I'm pretty new with all this, I'm sure I'm doing something wrong. But what am I missing?

bernhard-42, 2020-11-20:
This is CadQuery related, and I actually don't know why this doesn't work. Looks like the API changed. 
However, you could use `result.val().exportStl("test.stl", 0.1)`

bernhard-42, 2020-11-20:
Alternatively, leveraging `jupyter-cadquery`

```
from jupyter_cadquery import exportSTL
exportSTL(Part(result), "test.stl")
```

sergi, 2021-01-11:
Thanks a lot for this, it worked perfectly. Appreciate your help!

closed 2021-01-11 by sergi