bernhard-42/jupyter-cadquery #30: Elementary installation error

amca01, 2020-11-22:
I am using Windows 10, and following the instructions to the letter.  However, the command

    jupyter-labextension install --no-build $(cat labextensions.txt)

produces the error 

    ValueError: "$(cat" is not a valid npm package

I am not very knowledgeable about npm in general; all I want is to be able to use cadquery, and hopefully within jupyter.  I ended up going through each of the four lines in `labextensions.txt` one by one - but is there a better way?

bernhard-42, 2020-11-22:
Apologies, @amca01 , this is a UNIX command - it never came to my mind that this doesn't work on Windows.
I expect `jupyter-labextension` works under Windows, so please try 

```
jupyter-labextension install --no-build @jupyter-widgets/jupyterlab-manager@2.0.0 @jupyter-widgets/jupyterlab-sidecar@0.5.0 jupyterlab-datawidgets@6.3.0 jupyter_cadquery@v1.0.0
```

These are the strings in the `labextensions.txt` file and the syntax with `cat` is short way under UNIX to achieve the long command above.

amca01, 2020-11-22:
No need to apologise!  It's just that in my mind whenever I see "conda" I assume a Windows system - I use pip generally on my Linux box.  However, for this year what with working from home, I have been switching almost completely to Windows for everything (not without some annoyances).  Mostly things work fine.  And in the end I did exactly what you recommended, just in four separate commands.  The main thing I miss on Windows is a decent shell, like bash or zsh on Linux.

Anyway, I have got cadquery working very nicely in Jupyter, and I like it very much - thank you for your hard work.  I will see how easy it is to switch from OpenJSCAD, which has been my scripting CAD up until now.

Thank you again.

bernhard-42, 2021-02-28:
@amca01 I've release the release candidate of the next version. It supports CadQuery 2.1 and (relevant for this issue) Jupyter Lab 3.0 which simplifies the installation drastically and removes the need of labextension installations (see README.md)

bernhard-42, 2021-03-06:
@amca01 Just released 2.0.0 which does not need this step any more

amca01, 2021-03-07:
Thank you very much indeed for both your messages!  Lest you think me
unpardonably rude for not replying earlier, I have been busy both with the
start of the academic teaching semester, and the sudden debility of my
mother due to some strokes, and her admittance to a nursing home.

I'll look forward to experimenting with cadquery on Linux as soon as I can.

regards and again thanks,
Alasdair

bernhard-42, 2021-05-15:
Closing, because the whole labextensions installation has been changed and this is (luckily) not needed any more

closed 2021-05-15 by bernhard-42