bernhard-42/jupyter-cadquery #35: Make a release with cadquery 2.1

fischman, 2021-02-14:
Now that cadquery [2.1 is stable](https://github.com/CadQuery/cadquery/blob/master/changes.md#21-stable-release) IWBN to have a release of jupyter_cadquery that uses this version.

bernhard-42, 2021-02-20:
@fischman It is planned. However, I am applying a lot of changes lately and I am thinking about supporting Jupyterlab 3 to get rid of all the npmjs quirks (see https://github.com/bernhard-42/jupyter-cadquery/issues/34), but not sure right now.
I guess it will take another few weeks.

There is a beta docker container from Dec 12th (https://github.com/bernhard-42/jupyter-cadquery#b-using-a-docker-image)
Would this be sufficient for the moment?

fischman, 2021-02-20:
@bernhard-42 thanks for the reply. Indeed I've been using the beta image (`bwalter42/jupyter_cadquery:2.0.0-beta3`) but the reason I filed this bug was a divergence in the Assembly API between the cadquery in that image and the 2.1 release. Details are in item 3) of https://groups.google.com/g/cadquery/c/_qIbSa2wMXc

As a newcomer to cadquery, many of the things I try don't work for the first N times I try, and now that I know about this difference from 2.1dev to 2.1 it makes me wonder in each of my attempts whether this is me doing it wrong (~always the case) or a bug fixed in the 2.1 final release whose fix is missing from the beta image (hasn't happened yet except for the assembly issue above). So I'd like a final 2.1 cadquery in a jupyter_cadquery docker image if only for the sake of silencing those doubts :)

bernhard-42, 2021-02-28:
@fischman I published the release candidate with CadQuery 2.1 and Jupyterlab 3 support. As usual, you can try it with 
`docker run -it --rm -v $WORKDIR:/home/cq -p 8888:8888 bwalter42/jupyter_cadquery:2.0.0-rc1`

fischman, 2021-02-28:
@bernhard-42 thanks, seems to work well.
One issue I noticed is that (CQ) assemblies don't get automatically shown (see second cell), although calling `show` on them does work well (see third cell):
[image]

jmwright, 2021-02-28:
>  I published the release candidate with CadQuery 2.1 and Jupyterlab 3 support. 

@bernhard-42 Thanks for all your work on this! jupyter-cadquery is an important part of the CadQuery ecosystem, and it's great to see this progress.

bernhard-42, 2021-03-01:
@fischman Nice catch. I am so used to using `show` in the meantime to control all the flags for the renderer that I missed this. Will fix it in the final version.

What OS and browser are you using? There seems to be more space used for scrollbars in the above screenshot than for me on a Mac

bernhard-42, 2021-03-01:
@jmwright Glad to see the tool is used! Btw. not sure you've spotted it, I've added the version of CadQuery and jupyter-cadquery to the info box. So it should be clear now from a screenshot, what the underlying Cadquery version is, especially in case of problems

fischman, 2021-03-01:
@bernhard-42 this is in chrome (stable channel, I think, can check tomorrow) on os/x big sur.

bernhard-42, 2021-03-06:
@fischman Just released 2.0.0. Assemblies (both types) now show automatically

fischman, 2021-03-06:
@bernhard-42 confirmed. Thanks!

closed 2021-03-06 by fischman