bernhard-42/jupyter-cadquery #38: Integrate with conda-forge

bernhard-42, 2021-02-28:
Add jupyter-cadquery to conda-forge

bernhard-42, 2021-02-28:
see some details can be found in https://github.com/bernhard-42/jupyter-cadquery/issues/34 with @bollwyvl

bollwyvl, 2021-03-06:
Some things while digging around:
- the hard (e.g. `==`) pins in `install_requires` are pretty hard for downstream maintainers over time
  - a bottom pin is always _great_, but at least having a top pin, which is reviewed prior to release, helps give some wiggle room, as you can at least lobby the upstream for a point release with a backport-in-hand of e.g. security fixes
  - using `tox` to test against a range of python/runtime versions is the gold standard e.g. 
    - the oldest viable environment
    - the highest viable environment
    - an entirely unpinned version (potentially with `--pre` enabled) to help detect upcoming issues 
- having the `jupyterlab` pin in both places makes things a little more fragile
  - potentially just have the `-widgets` package carry that dependency
- looks like the `nbextensions` are now handled by the `jupyter-cadquery-widget` package, but are still listed in `data_files`
- not having a `cadquery` version pin seems like it will be problematic over time, even if it's not possible to install from pypi
  - since it's _not_ installable from pypi, one can at least put it behind an `extras_require` so that the version required/tested is at least _documented_ somewhere. 
    - this can be used by downstreams to test, though not through "normal means" like `pip check` (which ignores `extras` anyway)

Again, still no closer to having you your upstreams on `conda-forge`... the test cycle is quite slow, and I'm somewhat shooting in the dark on some of these things.

bernhard-42, 2021-03-06:
> the hard (e.g. ==) pins in install_requires are pretty hard for downstream maintainers over time

Happy to change
`"webcolors==1.11.1"`
`"sidecar==0.5.0"`
to
`"webcolors~=1.11"`
`"sidecar~=0.5"`
if that helps

> having the jupyterlab pin in both places makes things a little more fragile

Agreed, I can remove it from juypter-cadquery

> looks like the nbextensions are now handled by the jupyter-cadquery-widget package, but are still listed in data_files

That's a bug, will remove it

> not having a cadquery version pin seems like it will be problematic over time

Happy to add cadquery 2.1 to extra requires. I didn't need it up to now since it is not on pypi as you said

If I do these changes, would I then need to deploy both packages with newer versions (2.0.3 for jupyter-cadquery-widgets and 2.0.1 for jupyter-cadquery) or would it be sufficient to have the changes in master?

bollwyvl, 2021-03-06:
since i can't package your stuff _anyway_ yet, no need to publish!

bernhard-42, 2022-02-25:
@shimwell What would be your requirements? 
jupyter-cadquery plays well with conda as long as it is provided in the pip section of the environment yaml file.

One of the biggest issues I have is that CadQuery doesn't have versions or tags. In order to participate on features like `Sketch` I have to recommend to install cadquery=master. How to you solve this iswsue?

shimwell, 2022-02-25:
I am able to install into a conda environment nicely

However I am quite keen to build a conda package that makes use of jupyter-cadquery and I think that when making a conda package all dependencies must also be available on conda. Therefore I can't include jupyter-cadquery in the package I am currently making. The meta.yml for my package is [here](https://github.com/fusion-energy/paramak/blob/cec738b278c285a17eaa69fc1f35ea4788204a8c/conda/meta.yaml#L35) and you can see I have jupyter-cadquery commented out.

So currently I have a [two part](https://paramak.readthedocs.io/en/main/install.html#install-conda) install process which is conda install my package (this brings in cadquery and others) then pip install jupyter_cadquery.

I guess on the cadquery front that lack of tag will be solved once cadquery 2.2 is released. I wonder if in theory jupyter-cadquery could be published to either conda-forge channel, cadquery channel or another channel?

bernhard-42, 2022-02-25:
I have never built a conda package, so I first need to understand how to do so.
And I need to build conda-forge packages for all three sub projects I guess. I need to check the process ...

> I guess on the cadquery front that lack of tag will be solved once cadquery 2.2 is released. I wonder if in theory jupyter-cadquery could be published to either conda-forge channel, cadquery channel or another channel?

btw. my problem around cadquery=master is for the docker container. I artificially need to change something else in my Dockerfile in order to pick up the latest CadQuery master. In general I would prefer to pin the version to a version I have tested it on. But, no tags exist ...

bernhard-42, 2022-02-25:
I will also take inspiration from your conda packages and meta.yaml files ...

shimwell, 2022-02-25:
I recall there was a reason that CadQuery2 is in a dedicated channel and noticed[ this issue](https://github.com/CadQuery/cadquery/issues/287) that might be of interest

I think conda-forge packages need all their dependencies to be on conda forge so perhaps I should ask this question over on cadquery repo first

bernhard-42, 2022-02-25:
Good point. 
Could you please add me to your question there? I would like to follow and understand whether conda-forge would work. Thanks!

bernhard-42, 2022-06-19:
Since cadquery is available in pypi now, this is not necessary any more from my point of view

closed 2022-06-19 by bernhard-42