bernhard-42/jupyter-cadquery #41: Add ability to copy the replay selection as text

wcraigtrader, 2021-04-05:
I have a complex library (1000+ lines, several modules) that generates CadQuery models from specifications. This lets me write a small class that renders a large model, but sometimes this results in hard-to-debug code. The jupyter-cadquery replay() function is extremely useful for debugging, but it could be more useful if you could click a button and copy the actual calls and turn something like this:

[image]

Into the actual code:
```
import cadquery as cq

part = (cq.Workplane().rect(142.875,213.360)
        .workplane(offset=-30.1625).rect(142.875,213.260)
        .loft()
        .cut(
            cq.Workplane().union(
                cq.Workplane().rect(128.99127,195.66627)
                .workplane(offset=-9.535).rect(127.0,193.675)
                .loft()
                .edges('<Z').fillet(0.4)
                .translate((0,0,0.01)).translate((0,0,0))
            )
        )
        .faces('>Z').moveTo(-62.8625,-96.2).circle(0.875).cutThruAll()
        .faces('>Z').moveTo(-62.8625, 96.2).circle(0.875).cutThruAll()
        .faces('>Z').moveTo( 62.8625,-96.2).circle(0.875).cutThruAll()
        .faces('>Z').moveTo( 62.8625, 96.2).circle(0.875).cutThruAll()        
   )
```
(It doesn't have to be prettied up -- just extracting the functions and arguments into code would be wonderful.)

bernhard-42, 2021-04-05:
@wcraigtrader If you use `r = replay(obj)` then `r.stack` provides the content of the replay list, e.g.
```
[("box('10', '20', '30')", <cadquery.cq.Workplane at 0x7f5874502e80>),
 ("edges('>X or <X')", <cadquery.cq.Workplane at 0x7f58744c2a30>),
 ("chamfer('2') => box1", <cadquery.cq.Workplane at 0x7f5873994f70>),
 ("| box('8', '18', '28')", <cadquery.cq.Workplane at 0x7f587453a850>),
 ("| edges('>X or <X')", <cadquery.cq.Workplane at 0x7f587399af40>),
 ("| chamfer('2') => box2", <cadquery.cq.Workplane at 0x7f58744b0550>),
 ("cut('box2')", <cadquery.cq.Workplane at 0x7f5874480fd0>),
 ('| transformed(offset=(0, 15, 7))',
  <cadquery.cq.Workplane at 0x7f5874452760>),
 ("| box('30', '20', '6')", <cadquery.cq.Workplane at 0x7f5874452bb0>),
 ("| edges('>Z')", <cadquery.cq.Workplane at 0x7f58744b02e0>),
 ("| fillet('3') => box3", <cadquery.cq.Workplane at 0x7f58739a3400>),
 ("cut('box3')", <cadquery.cq.Workplane at 0x7f58744b09a0>),
 ('| box3', <cadquery.cq.Workplane at 0x7f58739a3400>),
 ("| mirror('XY')", <cadquery.cq.Workplane at 0x7f58744b0b50>),
 ("| translate('(0, -5, 0)') => box4",
  <cadquery.cq.Workplane at 0x7f58744b07c0>),
 ("cut('box4')", <cadquery.cq.Workplane at 0x7f58744b0790>)]
```

Would that already help?
Of course it is no code, but it could be parsed I guess

bernhard-42, 2021-04-05:
and, easier to use, `r.to_array(obj)` returns a list of Step objects, e.g. Step object 0:
```
{'level': 0,
 'func': 'box',
 'args': (10, 20, 30),
 'kwargs': {},
 'var': '',
 'result_name': '',
 'result_obj': <cadquery.cq.Workplane at 0x7f5874502e80>}
```

bernhard-42, 2021-05-15:
Closing for now. If you have any further questions, please reopen

closed 2021-05-15 by bernhard-42