bernhard-42/jupyter-cadquery #42: Preview: Juypter CadQuery now has a Viewer component for debugging in IDEs

bernhard-42, 2021-05-06:
This is a beta version of the IDE CadQuery viewer I talked about earlier. If you want to give it a try, happy to get feedback.

**Installation**

```bash
pip install git+https://github.com/bernhard-42/jupyter-cadquery.git@viewer2
```

**Starting** (on Mac, Linux and Windows)

```bash
jcv
```

[image]

As a **simple test** add the following code to `test.py`:

```python
import cadquery as cq
from jupyter_cadquery.viewer.client import show

box = cq.Workplane().box(1, 1, 1).faces(">Z").hole(0.6).edges().fillet(0.05)
show(box, axes=True, transparent=True)
```

```bash
python test.py
```
    
[image]

Jojain, 2021-05-06:
Thanks for the work it looks amazing.

I've tried to run it by typing `jcv` in fresh conda env with only cadquery and your branch but it fails to launch. I get an import error for `import win32api` with the full traceback : 
```
(jcv) C:\Users\Romain>jcv
[Voila] Using C:\Users\Romain\AppData\Local\Temp to store connection files
[Voila] Storing connection files in C:\Users\Romain\AppData\Local\Temp\voila_rhl03yp0.
[Voila] Serving static files from d:\programmes\miniconda\envs\jcv\lib\site-packages\voila\static.
[Voila] The port 8866 is already in use, trying another port.
[Voila] Voilà is running at:
http://localhost:8867/
[Voila] WARNING | Notebook viewer.ipynb is not trusted
[Voila] WARNING | Could not find a kernel named 'jvc', will use  '597585c0-bc20-4126-936a-8287d64ef80b'
ERROR:tornado.application:Uncaught exception GET / (::1)
HTTPServerRequest(protocol='http', host='localhost:8867', method='GET', uri='/', version='HTTP/1.1', remote_ip='::1')
Traceback (most recent call last):
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\tornado\web.py", line 1704, in _execute
    result = await result
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\voila\handler.py", line 154, in get
    async for html_snippet, resources in self.exporter.generate_from_notebook_node(notebook, resources=resources, extra_context=extra_context):
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\voila\exporter.py", line 100, in generate_from_notebook_node
    async for output in self.template.generate_async(nb=nb_copy, resources=resources, **extra_context, static_url=self.static_url):
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jinja2\asyncsupport.py", line 35, in generate_async
    yield self.environment.handle_exception()
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jinja2\environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jinja2\_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "d:\programmes\miniconda\envs\jcv\share\jupyter\voila\templates\lab\index.html.j2", line 4, in top-level template code
    {% from 'voila_setup.macro.html.j2' import voila_setup with context %}
  File "d:\programmes\miniconda\envs\jcv\share\jupyter\nbconvert\templates\lab\index.html.j2", line 3, in top-level template code
    {% from 'jupyter_widgets.html.j2' import jupyter_widgets %}
  File "d:\programmes\miniconda\envs\jcv\share\jupyter\nbconvert\templates\lab\base.html.j2", line 224, in top-level template code
    {% set div_id = uuid4() %}
  File "d:\programmes\miniconda\envs\jcv\share\jupyter\nbconvert\templates\base\display_priority.j2", line 1, in top-level template code
    {%- extends 'base/null.j2' -%}
  File "d:\programmes\miniconda\envs\jcv\share\jupyter\nbconvert\templates\base\null.j2", line 26, in top-level template code
    {%- block body -%}
  File "d:\programmes\miniconda\envs\jcv\share\jupyter\nbconvert\templates\base\null.j2", line 29, in block "body"
    {%- block body_loop -%}
  File "d:\programmes\miniconda\envs\jcv\share\jupyter\voila\templates\lab\index.html.j2", line 55, in block "body_loop"
    {%- block body_loop -%}
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jinja2\asyncsupport.py", line 174, in auto_await
    return await value
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\voila\handler.py", line 166, in _jinja_kernel_start
    kernel_id = await ensure_async(self.kernel_manager.start_kernel(
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\nbclient\util.py", line 85, in ensure_async
    result = await obj
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jupyter_server\services\kernels\kernelmanager.py", line 186, in start_kernel
    kernel_id = await ensure_async(self.pinned_superclass.start_kernel(self, **kwargs))
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jupyter_server\utils.py", line 176, in ensure_async
    result = await obj
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jupyter_client\multikernelmanager.py", line 483, in start_kernel
    await fut
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jupyter_client\multikernelmanager.py", line 460, in _add_kernel_when_ready
    await kernel_awaitable
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jupyter_client\manager.py", line 657, in start_kernel
    kernel_cmd, kw = self.pre_start_kernel(**kw)
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jupyter_client\manager.py", line 286, in pre_start_kernel
    self.write_connection_file()
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jupyter_client\connect.py", line 466, in write_connection_file
    self.connection_file, cfg = write_connection_file(self.connection_file,
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jupyter_client\connect.py", line 136, in write_connection_file
    with secure_write(fname) as f:
  File "d:\programmes\miniconda\envs\jcv\lib\contextlib.py", line 117, in __enter__
    return next(self.gen)
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jupyter_core\paths.py", line 461, in secure_write
    win32_restrict_file_to_user(fname)
  File "d:\programmes\miniconda\envs\jcv\lib\site-packages\jupyter_core\paths.py", line 387, in win32_restrict_file_to_user
    import win32api
ImportError: DLL load failed while importing win32api: Le module spécifié est introuvable.
ERROR:tornado.general:Cannot send error response after headers written
WARNING:tornado.general:403 GET /voila/files/favicon.ico (::1): File not whitelisted
WARNING:tornado.access:403 GET /voila/files/favicon.ico (::1) 1.99ms
``` 
Note that on launch it triggers my avast antivirus (which say that it's safe to run but still I don't know if that's expected)

bernhard-42, 2021-05-06:
You could also just start `jupyter lab` in the environment where the new version is installed and paste the two lines 

```bash
from jupyter_cadquery.viewer.server import start_viewer
start_viewer()
```
into a cell and execute it (voila just automates that)
Does it find `start_viewer`?

bernhard-42, 2021-05-06:
A simpler test whether the installation worked would be 

```
(jcv) C:\Users\bernhard>python
Python 3.8.8 (default, Apr 13 2021, 15:08:03) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.executable
'C:\\Users\\bernhard\\miniconda3\\envs\\jcv\\python.exe'
>>> from jupyter_cadquery.viewer.server import start_viewer
>>>
```

I am surprised that in your case a kernel 597585c0-bc20-4126-936a-8287d64ef80b is selected. Usually it should be `python3` in a fresh environment. This is how it looks like for me:

```
(jcv) C:\Users\bernhard>jcv
[Voila] Using C:\Users\bernhard\AppData\Local\Temp to store connection files
[Voila] Storing connection files in C:\Users\bernhard\AppData\Local\Temp\voila__rfmrk1o.
[Voila] Serving static files from c:\users\bernhard\miniconda3\envs\jcv\lib\site-packages\voila\static.
[Voila] Voilà is running at:
http://localhost:8866/
[Voila] WARNING | Notebook viewer.ipynb is not trusted
[Voila] WARNING | Could not find a kernel named 'jvc', will use  'python3'
[Voila] Kernel started: 931fa6fa-c4a3-4ed0-adb3-4902a10b3d4e
WARNING:tornado.general:403 GET /voila/files/favicon.ico (::1): File not whitelisted
WARNING:tornado.access:403 GET /voila/files/favicon.ico (::1) 1.00ms
WARNING:tornado.general:403 GET /voila/files/voila/jupyter-threejs.js (::1): File not whitelisted
WARNING:tornado.access:403 GET /voila/files/voila/jupyter-threejs.js (::1) 0.00ms
WARNING:tornado.general:403 GET /voila/files/voila/jupyter_cadquery.js (::1): File not whitelisted
WARNING:tornado.access:403 GET /voila/files/voila/jupyter_cadquery.js (::1) 0.00ms
```

Jojain, 2021-05-06:
It works ! 
I've tried it with a more complex example and everything seems to run smoothly.
If I call `show` several times it refresh the view and doesn't populate the view with several parts. I don't know if it's intended or if it's work in progress.

bernhard-42, 2021-05-07:
Nice! I will look into voila and its kernel issue, since that would be easier.

`show` works differently than `show_object` in CadQuery. You can give more than one object: `show(obj1, obj1, ...)`. You can them in the view hide/show parts to better understand complex objects. If you use PartGroup and Parts from jupyter cadquery, you can also build hierarchies of objects and name and color them for the view (see an example in the Readme)

Would that help?

jmwright, 2021-05-07:
@bernhard-42 I'm not understanding to how execute code. I have the viewer open, and you say above to add code to test.py. It looks in the gif like you're adding code in VS Code (maybe?), so I'm not sure how to make that work. Where is test.py located?

Jojain, 2021-05-07:
@jmwright I think you can run the code test.py from wherever you want, if I understand it well, when you run your python script the show function send the data to the ipython kernel so it's displayed in jupyterlab.

For my case I got it working by running test.py from vscode in the conda env where jcv is installed after having started jupyterlab as explained by @bernhard-42

bernhard-42, 2021-05-07:
@Jojain correct.
@jmwright 
The `show` function imported will tessellate the objects provided and then send the numpy arrays via `zmq` to the viewer. So you can call `python test.py` from the command line or run it in VSCode or any other editor from anywhere. You only need to use the python interpreter of the conda env you've installed the viewer into. 
The viewer is stateless, so you can even send (show) objects from different editor instances.
If you set `reset_camera=False`, different calls to `show` will use the same angle to look at the object.

So basically, when the voila thing works (see below) you don't even see jupyter. And restarting the viewer simply is a browser page refresh.

This also means that there is no dependency for the viewer part on Cadquery. The viewer only deals with numpy arrays and pythreejs. 
Only the show function uses Cadquery code. And if jupyter cadquery is deployed to the same env as your cadquery, the versions will fit automatically.

Additionally, I found the issue around `voila`: I need to create a kernelspec called `jcv` first and tell `voila` to use it. Will update the git repo today or tomorrow. Up to then `jupyter lab` will do the trick

jmwright, 2021-05-07:
Ah, ok. It's working now for me. That's mind warping for me to run Python on the command line and have it update in the browser.

Nice work! I think that once you release this integration I'll put a note at the top of the [cadquery-gui](https://github.com/jmwright/cadquery-gui) readme that this is the best replacement for that project since I really only do security updates on it now. I'm assuming this will display assemblies and do animations as well, correct?

bernhard-42, 2021-05-07:
Yes, it can show everything jupyter-cadquery can - except animations for now (see below)

A little more details:
 
- The first step is to display the UI

- The second step is actually a pipeline:
  - convert what so ever cadquery artifact to internal PartGroup/Part hierarchy (including cadquery assemblies)
  - tessellate every object in PartGroup hierarchy and build a dict with all numpy arrays
  - then
    - In Jupyter this dict gets rendered and the result gets pushed into the view
    - In the viewer the show function sends the dict via zmq to the server part of the UI. There the dict will be rendered and visualised in the same way as in Jupyter

This ensures that whatever I build into jupyter-cadquery will also be available for the viewer. I didn't want to maintain 2 code streams ...

For the time being, there is no message defined to push animation data to the viewer. This is some future feature I will consider in the next release. This release will cover everything except animations.

bernhard-42, 2021-05-15:
Release RC1 of Jupyter-CadQuery, so closing this issue

closed 2021-05-15 by bernhard-42