bernhard-42/jupyter-cadquery #43: Resize view

Jojain, 2021-05-10:
Hello, 
I've tried to resize the 3D view in your new viewer that you presented here #42 , but it don't know how to do it.
I tried to change the size width like that : 
`show(shapes, cad_width = 1000)`
But it doesn't do anything, I've also tried it with regular jupyter-cadquery and it doesn't do anything either. I am doing it wrong ?

Jojain, 2021-05-11:
Since voilà didn't worked for now I use the viewer by starting jupyter lab manually and to set the view this works : 
```
from jupyter_cadquery.viewer.server import start_viewer
from jupyter_cadquery.defaults import set_defaults
set_defaults(cad_width=1400, height=800)
start_viewer()
```
Thanks !
Note that in issue #42 you told me that I could call `show(obj1, ob2)` but it's actually not possible since the `show` function you defined in the `viewer` doesn't accept multiple objects.  However packaging every shape in a PartGroup works well and is nice.

PS : That's unrelated and not a big deal but in case you didn't know the jupyter-cadquery view icons don't look good on dark background of darkmode jupyter lab, see below :
[image]

bernhard-42, 2021-05-13:
@Jojain There is a new version available, see https://github.com/bernhard-42/jupyter-cadquery/issues/42#issue-877159979

Notable features / fixes:
- `jcv.cmd` has now the logic to create a juypter kernelspec `jcv` which is expected by the viewer.  It now should work under windows.
- The installation of `pywin32` is mandatory on windows. It didn't work with pip, however, `conda install pywin32` solves the issue
- The parameters `cad_width` and `height` of the `show` function now resize the cad view
- `jcv.cmd` comes with parameters for cad_width (`-w`) and height (`-h`). Note, if you add specific width and height, you need to `set_defaults` in your python code to the same values, since the client defaults will always win.
- There is an initial dark mode. Need to tweak the CSS for some elements (e.g. scroll bars a black on dark grey), but it already works: `jcv -d` 
- `show` now takes more than one object
- `show_object` now shows all objects incrementally
- Jupyter-CadQuery viewer now has a logo :-)

I should have added quite some of the stuff from your feedback. Maybe you can give it a try whether it now works better?

If it does, I will soon make a release candidate out it to get out of the "pip install from git" woods ...

bernhard-42, 2021-05-15:
RC1 released ...

bernhard-42, 2021-05-15:
@Jojain Does it work? 
Can we close this issue?

Jojain, 2021-05-15:
I cannot try it before Monday , I will let you know then if it works

Jojain, 2021-05-17:
I tried it and it's all good, resizing from command line or overriding the settings with the `show` function works !

closed 2021-05-17 by Jojain