bernhard-42/jupyter-cadquery #44: Jupyterlab is non-responsive for some time after rendering 3D model

TormodLandet, 2021-05-14:
Every time I show a model in Jupyter lab the following happens:
1) The model is rendered part by part until complete
2) The notebook is then non-responsive while Jupyter lab's bottom bar rapidly flickers between Busy and Idle for some time (maybe 10-20 seconds, more for larger models?)

Step number two seems to take more time the more objects are in the scene. The model is already rendered, so it seems like a bug, but maybe it does something that is necessary? Step two seems to take more time than step 1.

I'm using Firefox on Ubuntu, Jupyterlab 3.0.13 from conda (I followed the description in the jupyter-cadquery readme). I just upgraded to jupyter-cadquery-2.2.0rc0 to see if the problem was fixed.

#### Minimal example that triggers this in jupyterlab:

```python
from math import sin, cos, pi

import cadquery as cq
import jupyter_cadquery.cadquery as jcq

xy = cq.Workplane("XY")
test = cq.Assembly(name="Test assembly")

r = 3
d = 0.1
for i in range(0, 361, 10):
    x = r * cos(i * 180 / pi)
    y = r * sin(i * 180 / pi)
    box = xy.box(d, d, r/2)
    box = box.translate([x, y, 0])
    test.add(box)

jcq.show(test)
```

bernhard-42, 2021-05-14:
@TormodLandet 
Could it be that you are running out of memory on your machine? I had similar behaviour on my 8GB MacBook with some models.
I just gave your code a try on my big Linux server and it rendered without any busy flickers in 1.5 sec.

TormodLandet, 2021-05-14:
I have 16 MB of RAM. While executing the cell the laptop is hovering around 50% utilization without budging much (a couple of percent maximum). The CPU has a peak on one of the cores for 2 seconds (rendering) and then for about 20 seconds on another core (busy flickering). I tried triangulating the model and saving to file to see if this was the culprit, but that is very fast, so it must be some interaction with the finished rendering and the nodebook. The view of the triangulated model truly seems "ready" before the busy flickers happen.

Can I turn off all GUI (buttons, tree etc) to just test the 3D rendering isolated?

Btw, is there a better way to triangulate an assembly than this? I did not find a better way when I was debugging this problem.

```python
from cadquery.occ_impl.exporters import toCompound

def triangulate(assembly: cq.Assembly, tolerance:float=0.1, angularTolerance:float=0.1):
    """
    Trianulate an assembly, returns
    - vertices: a list of 3-tuples (float, coordinates)
    - indices: list of 3-tuples (int, node indices for each triangle)
    - shape_ids: list of integers, one for each vertex. Can be used
      to paint each shape with a different color etc
    """
    all_vertices, all_indices, all_ids = [], [], []
    nshapes = nnodes = 0

    def triangulate_subpart(part):
        nonlocal nshapes, nnodes, all_vertices, all_indices, all_ids
        
        # Triangulate this assembly
        try:
            compound = toCompound(part.obj)
            vertices, indices = compound.tessellate(tolerance, angularTolerance)
            ids = [nshapes] * len(vertices)
            vertices = [(v.x, v.y, v.z) for v in vertices]
            indices = [[node + nnodes for node in cell] for cell in indices]

            all_vertices.extend(vertices)
            all_indices.extend(indices)
            all_ids.extend(ids)
            nshapes += 1
            nnodes += len(vertices)
        except AttributeError:
            pass

        # Triangulate any children
        for child in part.children:
            triangulate_subpart(child)
    
    triangulate_subpart(assembly)
    
    return all_vertices, all_indices, all_ids

def save_with_meshio(assembly: cq.Assembly, filename):
    """
    Save the assembly (triangulated) to one of the formats supported
    by meshio (file names ending in .vtk end up in that format etc)
    """
    import meshio
    
    
    points, indices, ids = triangulate(assembly)
    cells = [("triangle", indices)]
    data = {'shape': [float(shape_id) for shape_id in ids]}
    meshio.write_points_cells(filename, points, cells, point_data=data)

save_with_meshio(test, "test.vtk")  # Can be opened in ParaView etc
```

bernhard-42, 2021-05-14:
> Can I turn off all GUI (buttons, tree etc) to just test the 3D rendering isolated?

Try `jcq.show(test, tools=False)`

> I tried triangulating the model and saving to file to see if this was the culprit, but that is very fast

You could try `jcq.show(test, tools=False, timeit=True)`
It provides a lot of timing infos

bernhard-42, 2021-05-14:
> Btw, is there a better way to triangulate an assembly than this?
Not sure there is another way. I found the tessellator of CadQuery too slow, so this is my approach per object: https://github.com/bernhard-42/jupyter-cadquery/blob/3b6c3d435b59f554c0ec34cbf2fdd6df9cd360e6/jupyter_cadquery/tessellator.py#L83
Additionally, for nice visualization, one also needs the normals ...

TormodLandet, 2021-05-14:
I tried `jcq.show(test, tools=False, timeit=True)`. It prints a lot of info that ends with  `1.425 sec: overall`, but this is printed before the busy-flickering starts

bernhard-42, 2021-05-14:
I just got mine flickering: I opened the same notebook in Firefox and in Chrome at the same time.
Now JuypterLab in both browsers was flickering quite some time. Same happens when I have the same notebook open in Firefox in two windows.

Do you by chance have the same notebook open twice?

bernhard-42, 2021-05-14:
> It prints a lot of info

yep, one can reduce it by providing the level, i.e. `timeit=1` ... `timeit=5`. It was my debugging tool to develop the fastest possible tessellator ...

TormodLandet, 2021-05-14:
Aha, that made me think about extensions. I disabled all Firefox extensions and then it worked. It turns out that a library-access extension I have from a university was casing this (https://www.leanlibrary.com/download/). It probably inspects all connections or something (to see if they need tunneling through a login server to get access to paywalled academic papers without using VPN), and this has some unfortunate effects in this case ... :-( 

Sorry for bothering you!

closed 2021-05-14 by TormodLandet

bernhard-42, 2021-05-14:
No worries, glad it works now!