bernhard-42/jupyter-cadquery #45: Cad view controls

Jojain, 2021-05-18:
Is there a way to modify the controls of the cadviewport ?
By this I mean :
- Changing the keybinds/mousebinds of translation, rotation and zoom in the view
- Changing the sensibility of translation, rotations and zoom

Almost every CAD software use a different set of key/mouse bindings to navigate the 3D view so it's often frustrating when you don't have a consistent set across all of them

bernhard-42, 2021-05-19:
> Changing the sensitivity of translation, rotations and zoom

While not nice, one can do this now via:

```
d = show( ... )
d.cq_view.renderer.controls[0].rotateSpeed = 3
d.cq_view.renderer.controls[0].panSpeed = 5
d.cq_view.renderer.controls[0].zoomSpeed = 2
```

The `panSpeed` (translation) depends on the size of the object and it will currently be calculated as `(bb.xsize + bb.ysize + bb.zsize) / 300` with `bb`  being the bounding box of of the object.

I could add these as parameters to the default system, so one can set them via `set_defaults`

bernhard-42, 2021-05-19:
> Changing the keybinds/mousebinds of translation, rotation and zoom in the view

Well, threejs can since r99 (see https://stackoverflow.com/a/54154920 ), however pythreejs is stuck with r97 ( https://github.com/jupyter-widgets/pythreejs/blob/150ff1c10c868b17fefa63d19153b5ee1fe87f66/js/package.json#L40)

Opened https://github.com/jupyter-widgets/pythreejs/issues/359 for this.