bernhard-42/jupyter-cadquery #47: Unable to see "cadquery.Assembly" object in viewer

rohitramu, 2021-07-13:
I'm using the standalone viewer by running `jcv -d` before running my Python files locally.  I seem to be unable to show assembly objects.  I see the following console output, but no exceptions are thrown:
```
. sending ... 
 AttributeError: 'NoneType' object has no attribute 'max_dist_from_center'
```

Here's a minimal repro:
```python
import cadquery as cq
import jupyter_cadquery.viewer.client as j_viewer_client

box = cq.Workplane().rect(20, 20).extrude(40)
assembly = cq.Assembly().add(box)
j_viewer_client.show(assembly)
```

I've stepped through the code, and it seems like the jcv server is returning the above error message.  I found the message by inpecting the reply from the jcv server at this line:
https://github.com/bernhard-42/jupyter-cadquery/blob/a1b524aa17de231728a44155da3eb8e3e02edeb0/jupyter_cadquery/viewer/client.py#L59

Am I doing something wrong in my minimal repro?  Is there a problem in the jcv server?  Or maybe there's a problem in the client's convert() method?
https://github.com/bernhard-42/jupyter-cadquery/blob/a1b524aa17de231728a44155da3eb8e3e02edeb0/jupyter_cadquery/viewer/client.py#L83

rohitramu, 2021-07-13:
Looks like the only references to the `max_dist_from_center()` method is in this file:
https://github.com/bernhard-42/jupyter-cadquery/blob/8d9e7488280acc900c4652869056da8af91766b3/jupyter_cadquery/cad_view.py#:~:text=def%20_scale(self,return%20new_vec

Seems like the `bb` variable (i.e. the bounding box) was never set.  Why would this be the case with Assembly objects but not regular Workplane or Shape objects?

bernhard-42, 2021-07-13:
@rohitramu Thanks for raising the issue. I looked into it and it is a bug:
I missed one test case, i.e. an empty top assembly as in your case:

```python
assembly = cq.Assembly().add(box)
```

What works is to have the box in the top assembly for the time being:

```python
assembly = cq.Assembly(box)
```

If you want a quick fix immediately, then add `return` to the line https://github.com/bernhard-42/jupyter-cadquery/blob/a1b524aa17de231728a44155da3eb8e3e02edeb0/jupyter_cadquery/cad_objects.py#L398 to your installed jupyter_cadquery in site-packages to get 

```python
            else:
                return c_bb(shape, bb)
        return bb
```

I will publish a fix asap, however, does the workaround (add the box to the top assembly) work for you?

rohitramu, 2021-07-13:
Thanks for getting back so quickly!  Yes, that workaround works for me :).

Will you be updating the 2.2.0 package, or will you be incrementing the version?

bernhard-42, 2021-07-24:
The fix will be part of the next version (2.2.1). Need to find a little bit of time to prepare, test and publish.

tgrosinger, 2021-10-06:
I'm hitting this bug as well, and the fix worked.

For my future reference, file is `/opt/conda/envs/cq/lib/python3.8/site-packages/jupyter_cadquery/cad_objects.py` when running in the Docker container.