bernhard-42/jupyter-cadquery #53: scipy needed for installation

jchidley, 2021-09-25:
I've just done a fresh install on my Windows 10 machine and I had to install `scipy` to get `jcv` to work.

bernhard-42, 2021-09-25:
@jchidley The assumption was that jupyter-cadquery needs cadquery to work and scipy was a dependency of cadquery. So if you had installed jupyter-cadquery after cadquery, it was ok. However, it looks like cadquery dropped scipy as a dependency on master. Are you using the latest master (doesn't use scipy any more) or the release 2.1 (uses scipy)?

jchidley, 2021-09-25:
I am using the *master* version of `cadquery`, as installed by `mambaforge`

bernhard-42, 2021-09-25:
Thanks, good to know. 
I will try to remove the scipy dependency in the next release - I just use it for very few transformations, not worth keeping this dependency (as long as scipy was installed by CadQuery, this was no issue).