bernhard-42/jupyter-cadquery #55: Viewer mode from Docker

tgrosinger, 2021-10-05:
I'm attempting to run the `bwalter42/jupyter_cadquery:2.2.0` docker image, and connect to it in viewer mode from VS Code. I've been attempting to follow along with what's described in #40 and #42, but I'm having trouble determining what is still necessary now that it seems the docker image has been updated to a version which has the viewer mode built in.

When I exec into the container and run `python`  I get the following error.

```
~$ pythonPython 3.8.5 (default, Sep  4 2020, 07:30:14) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from jupyter_cadquery.viewer.client import show
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'jupyter_cadquery'
```

Please let me know if there are instructions I missed for starting in viewer mode, and then connecting from my IDE.
Thank you!

bernhard-42, 2021-10-05:
When working with the docker container, we need to simulate the `jcv` command since it is not ootb supported:

1) When you start the docker container also expose the zmq port 5555:

    ```

    run -it --rm -v $WORKDIR:/home/cq -p 5555:5555 -p 8888:8888 bwalter42/jupyter_cadquery:2.2.0
    ```
2) Open `http://127.0.0.1:8888/lab` in your browser

3) Create/open a notebook in Jupyter in the browser and run the following code in the first cell:

    ```
    from jupyter_cadquery.viewer.server import start_viewer
    start_viewer()
    ```

    Now you should see the cad viewer with the logo

4) In your IDE use the show or show_object command. It'll send the objects to port 5555 of the docker container and show the object in the jupyter cell. Note: There is no need to work in Jupyter, it is only the container for the viewer!

If you have a workdir for the docker command you can reuse this notebook in the next sessions.

The `jcv` command uses `voila` (a Jupyter viewer) to automate this for local installations.

tgrosinger, 2021-10-05:
Ah, this is very helpful. Thank you very much.

Just a little more context, for others that find this issue. I do not have conda or even Python installed on my computer. I first launch the docker container for `jupyter_cadquery`

```
docker run -it --rm -v $(pwd):/home/cq -p 8888:8888 bwalter42/jupyter_cadquery:2.2.0
```

Then I connect to that running container from VS Code using the "Remote - Containers" extension.

[image]

Open a web browser and setup the notebook as @bernhard-42 mentions above. Then in VS Code open a terminal (which will run in the container).

```
conda init bash
exit
```

Open another terminal

```
conda activate cq
python my_script.py
```

Your results should be rendered in the notebook!

I haven't yet gotten this hooked up to the debugger, but I believe at this point it should be possible.

tgrosinger, 2021-10-05:
I got the debugger working. First, install the Python VS Code extension. Be sure that you are installing it in the container.

Create `.vscode/settings.json` in your project and add the following:

```
{
	"python.pythonPath": "/opt/conda/envs/cq/bin/python"
}
```

Create `.vscode/launch.json` in your project and add the following (this might be optional):

```
{
	"version": "0.2.0",
	"configurations": [
		{
			"name": "Python: Current File",
			"type": "python",
			"request": "launch",
			"program": "${file}",
			"console": "integratedTerminal"
		}
	]
}
```

The Run and Debug buttons in VS Code should now work! And you can use them from any computer with Docker, no other dependencies.

closed 2021-10-05 by tgrosinger

bernhard-42, 2021-10-05:
Nice @tgrosinger and thanks for extending the context
I will take your recipe and add it to the docs in the next release

bernhard-42, 2021-10-08:
@tgrosinger I added a flag `-v` to the docker command which will start the viewer instead of JupyterLab, see https://github.com/bernhard-42/jupyter-cadquery#installation-and-usage section 2.
I thought that might make your workflow a little bit simpler