bernhard-42/jupyter-cadquery #57: 'exportSTL' import error

NormKOALA, 2021-10-31:
`ImportError: cannot import name 'exportSTL' from 'jupyter_cadquery' (C:\...\Anaconda3\envs\cadquery\lib\site-packages\jupyter_cadquery\__init__.py)`

Can someone help me to solve the problem?

NormKOALA, 2021-10-31:
Sorry, I understood what the problem was
Example is deprecated

bernhard-42, 2021-11-05:
@NormKOALA 
Should document it better, for simple objects you can simply use the CadQuery functions:

```python
import cadquery as cq
from cadquery import exporters

box = cq.Workplane().box(1,2,3)
exporters.export(box, "box.stl")
``` 

For `PartGroups` you could use:

```python
import cadquery as cq
from jupyter_cadquery.export import exportSTL
from jupyter_cadquery.cadquery import Part, PartGroup, show

box = cq.Workplane().box(10,20,30).edges().fillet(4)
sphere = cq.Workplane().sphere(12).translate((0,0,10))

quality = 0.01
angular_tolerance = 0.2

pg = PartGroup([Part(box),Part(sphere)])
exportSTL(
    pg,
    "box.stl", 
    tolerance=quality, 
    angular_tolerance=angular_tolerance
)
```

bernhard-42, 2022-03-13:
This is how it works now in Jupyter-CadQuery 3.0.0. Small change in imports. The Readme also reflects it

```python
import cadquery as cq
from jupyter_cadquery.export import exportSTL
from jupyter_cadquery import Part, PartGroup, show

box = cq.Workplane().box(10,20,30).edges().fillet(4)
sphere = cq.Workplane().sphere(12).translate((0,0,10))
pg = PartGroup([Part(box),Part(sphere)])

exportSTL(
    pg,
    "box.stl", 
    tolerance=0.01, 
    angular_tolerance=0.2
)
```

Closing for now. If you still have issues, please reopen.

closed 2022-03-13 by bernhard-42