bernhard-42/jupyter-cadquery #58: Error when starting from conda

flaviut, 2021-11-04:
I've installed this via miniconda if that's relevant, but I see the following when starting it:

```console
(jcq22) $ jcv -d
[JCV] Creating a Jupyter kernel specification called 'jcv' for this conda environment
[Errno 13] Permission denied: '/usr/local/share/jupyter'
[JCV] Copying the voila notebook to /home/user/etc/.jcv
[JCV] Signing the volia notebook
Notebook already signed: /home/user/etc/.jcv/viewer.ipynb
[JCV] Starting Jupyter CadQuery Viewer
[Voila] Using /tmp to store connection files
[Voila] Storing connection files in /tmp/voila_5q59dnvk.
[Voila] Serving static files from /home/user/etc/.conda/envs/jcq22/lib/python3.8/site-packages/voila/static.
[Voila] Voilà is running at:
http://tight-seat:8866/
[Voila] WARNING | Could not find a kernel named 'jcv', will use  'pypy3'
[Voila] Kernel started: 319c1075-9d09-42b7-9979-9c9e1cbc9367
...
ModuleNotFoundError: No module named 'jupyter_cadquery'
```

It seems like it's trying to install the thing system-wide, but doing that is bad practice on Linux (and probably other OSes) and requires privilege escalation.

Running `python -m ipykernel install --name jcv --display-name jcv --user`, which is the same as from the script, but with `--user` installs the kernel locally for the current user. This ends up working fine, since the launch script never checks for errors when it tries to create it again.

I'd suggest:

- using [`set -euxo pipefail` to make sure errors aren't ignored](https://vaneyckt.io/posts/safer_bash_scripts_with_set_euxo_pipefail/)
- creating the kernel locally, since there's no good reason to do it system-wide

bernhard-42, 2021-11-05:
@flaviut good point, I will add a flag to install the kernel as user.
From a bash perspective, I would agree with `-euo pipefile`, however, `-x` I'd like to avoid. No need to start `jcv` in bash debug mode

flaviut, 2021-11-06:
Sure, and thanks for looking into this!

My work is usually much less user facing, so I forgot about the `-x`. Skipping that sounds very logical :smile:.

flaviut, 2022-02-13:
That's great! Thank you!

closed 2022-02-13 by flaviut