bernhard-42/jupyter-cadquery #60: Docker image 2.2.1 out of date

qwelyt, 2022-01-19:
I seems that the latest docker image has an out of date CadQuery version. I tried to run my code with the docker image but got this

```
AttributeError: module 'cadquery.selectors' has no attribute 'AreaNthSelector'
```

This works in cq-editor.

Is it possible to update the image with the latest cadquery?

bernhard-42, 2022-02-13:
@qwelyt Sorry for the late reply, I was working on finalizing release 3.0. I have merge release3 today into master and 3.0rc3 is the last release candidate before the final release. It supports Sketch class and is available on docker hub

bernhard-42, 2022-03-13:
3.0.0 is on dockerhub now `docker run -it --rm -v $WORKDIR:/home/cq -p 8888:8888 bwalter42/jupyter_cadquery:3.0.0`
Hence, closing  the issue

closed 2022-03-13 by bernhard-42