bernhard-42/jupyter-cadquery #61: Missing import for sketches in 3.0rc1

KaruroChori, 2022-01-31:
Any attempt to include cq will result in the following log file:
[log.txt](https://github.com/bernhard-42/jupyter-cadquery/files/7967862/log.txt)

I followed the instructions reported on the README for branch 3 on a fresh install of ubuntu 21.
I was not able to find the docker image, so I cannot test if that environment is working properly.

bernhard-42, 2022-01-31:
@KaruroChori I just checked, the command given in the Readme installs cadquery 2.1 (which doesn't have Sketch)
Could you maybe try `conda create -n jcq3 -c conda-forge -c cadquery python=3.8 cadquery=master` to install the latest CadQuery with Sketch?

KaruroChori, 2022-01-31:
Yes that was it! Thank you!

bernhard-42, 2022-02-13:
@KaruroChori Just merged release3 into master. 3.0rc3 is the last release candidate before 3.0 (Readme is updated). If you find/found any issues, happy to fix before launching 3.0. Thanks

KaruroChori, 2022-02-13:
None so far, but I will keep you updated if I find any :).

bernhard-42, 2022-03-13:
closing for now. If there are still issues, please reopen

closed 2022-03-13 by bernhard-42