bernhard-42/jupyter-cadquery #63: Error displaying widget: model not found

altaic, 2022-02-15:
Hello,

I just upgraded to `jupiter-cadquery` 3.0.0rc3 from 3.0.0rc2 and I'm running into the error `Error displaying widget: model not found` whenever I attempt to call `jcq.show(...)` (after `import jupyter_cadquery as jcq`). When I upgraded to rc2, I ran into the same error, and IIRC I resolved it by upgrading various packages via `conda` and `pip`. However, I've tried upgrading various packages this time with no luck. Is there a reasonable way to debug this?

Perhaps related, I'm not sure why `conda info` shows `python version : 3.8.12.final.0` when this env was set up with `python 3.9.10` as can be seen in the `conda list` output.

Here is info about my conda and jupyter setup:
```
(py39) ➜  ~ conda info         

     active environment : py39
    active env location : /Users/xxxxxx/Applications/JupyterLab.app/Contents/Resources/jlab_server/envs/py39
            shell level : 2
       user config file : /Users/xxxxxx/.condarc
 populated config files : /Users/xxxxxx/Applications/JupyterLab.app/Contents/Resources/jlab_server/.condarc
          conda version : 4.11.0
    conda-build version : not installed
         python version : 3.8.12.final.0
       virtual packages : __osx=10.16=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /Users/xxxxxx/Applications/JupyterLab.app/Contents/Resources/jlab_server  (writable)
      conda av data dir : /Users/xxxxxx/Applications/JupyterLab.app/Contents/Resources/jlab_server/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/xxxxxx/Applications/JupyterLab.app/Contents/Resources/jlab_server/pkgs
                          /Users/xxxxxx/.conda/pkgs
       envs directories : /Users/xxxxxx/Applications/JupyterLab.app/Contents/Resources/jlab_server/envs
                          /Users/xxxxxx/.conda/envs
               platform : osx-64
             user-agent : conda/4.11.0 requests/2.26.0 CPython/3.8.12 Darwin/21.3.0 OSX/10.16
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

```
(py39) ➜  ~ jupyter labextension list
JupyterLab v3.2.9
/Users/xxxxxx/Applications/JupyterLab.app/Contents/Resources/jlab_server/envs/py39/share/jupyter/labextensions
        cad-viewer-widget v1.0.0 enabled OK (python, cad_viewer_widget)
        @voila-dashboards/jupyterlab-preview v2.1.1 enabled OK (python, voila)
        @jupyter-widgets/jupyterlab-manager v3.0.1 enabled OK (python, jupyterlab_widgets)
```

```
(py39) ➜  ~ conda list
# packages in environment at /Users/xxxxxx/Applications/JupyterLab.app/Contents/Resources/jlab_server/envs/py39:
#
# Name                    Version                   Build  Channel
anyio                     3.5.0            py39h6e9494a_0    conda-forge
appnope                   0.1.2            py39h6e9494a_2    conda-forge
argon2-cffi               21.3.0             pyhd8ed1ab_0    conda-forge
argon2-cffi-bindings      21.2.0           py39h89e85a6_1    conda-forge
asttokens                 2.0.5              pyhd8ed1ab_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
babel                     2.9.1              pyh44b312d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
black                     22.1.0             pyhd8ed1ab_0    conda-forge
bleach                    4.1.0              pyhd8ed1ab_0    conda-forge
brotlipy                  0.7.0           py39h89e85a6_1003    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2021.10.8            h033912b_0    conda-forge
cad-viewer-widget         1.0.0                    pypi_0    pypi
cadquery                  master                    py3.9    cadquery
cadquery-massembly        0.9.0                    pypi_0    pypi
[... 176 lines cut ...]
tomli                     2.0.1              pyhd8ed1ab_0    conda-forge
tornado                   6.1              py39h89e85a6_2    conda-forge
traitlets                 5.1.1              pyhd8ed1ab_0    conda-forge
typed-ast                 1.5.2            py39h89e85a6_0    conda-forge
typing_extensions         4.1.1              pyha770c72_0    conda-forge
typish                    1.9.3              pyhd8ed1ab_0    conda-forge
tzdata                    2021e                he74cb21_0    conda-forge
urllib3                   1.26.8             pyhd8ed1ab_1    conda-forge
utfcpp                    3.2.1                h694c41f_0    conda-forge
voila                     0.3.1              pyhd8ed1ab_0    conda-forge
vtk                       9.0.1           qt_py39h93184f3_210    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
webcolors                 1.11.1                   pypi_0    pypi
webencodings              0.5.1                      py_1    conda-forge
websocket-client          1.2.3              pyhd8ed1ab_0    conda-forge
websockets                10.1             py39h89e85a6_0    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
widgetsnbextension        3.5.2                    pypi_0    pypi
x264                      1!161.3030           h0d85af4_1    conda-forge
xz                        5.2.5                haf1e3a3_1    conda-forge
zeromq                    4.3.4                he49afe7_1    conda-forge
zipp                      3.7.0              pyhd8ed1ab_1    conda-forge
zlib                      1.2.11            h9173be1_1013    conda-forge
zstd                      1.5.2                h582d3a0_0    conda-forge
```

altaic, 2022-02-15:
I completely mangled my conda environment messing with this, so I remade both the 3.0.0rc2 and 3.0.0rc3 environments with the following conda yml files:

```
name: py39-jcqrc2
channels:
  - conda-forge
  - cadquery
dependencies:
  - python=3.9
  - pip
  - jupyterlab
  - cadquery=master
  - sympy
  - ipykernel
  - voila
  - pip:
    - jupyter-cadquery==3.0.0rc2
```

```
name: py39-jcqrc3
channels:
  - conda-forge
  - cadquery
dependencies:
  - python=3.9
  - pip
  - jupyterlab
  - cadquery=master
  - sympy
  - ipykernel
  - voila
  - pip:
    - jupyter-cadquery==3.0.0rc3
```

To be clear, I ran `conda env create --file py39-jcqrc2.yml` and `conda env create --file py39-jcqrc3.yml`. Then, in a jupyterlab notebook, I have:
```
import jupyter_cadquery as jcq
jcq.open_viewer('thetitle')
```
Running this using the `py39-jcqrc2` kernel*, it displays the cad viewer. However, running it using the `py39-jcqrc3` kernel*, it displays `Error displaying widget: model not found`.

\* I have `nb_conda_kernels` installed in the base environment, which makes all conda envs available as jupyterlab kernels.

altaic, 2022-02-15:
Looks like the issue is introduced in bernhard-42/cad-viewer-widget@f1d03ecb0cb5779d8b725f83fbd4a531eb32dc04. The previous commit works fine.

bernhard-42, 2022-02-15:
@altaic Thanks for the feedback.
I checked two things:
1) Mybinder for 3.0.0rc3, it works.
2) I copied your second yaml file, created an env and it works.

So, after installation, I see

```shell
$ jupyter-labextension list
JupyterLab v3.2.9
/opt/anaconda/envs/py39-jcqrc3/share/jupyter/labextensions
        cad-viewer-widget v1.0.0 enabled OK (python, cad_viewer_widget)
        @jupyter-widgets/jupyterlab-manager v3.0.1 enabled OK (python, jupyterlab_widgets)
        @voila-dashboards/jupyterlab-preview v2.1.1 enabled OK (python, voila)
```

When I start the browser and look into the browser console, I see version 1.0.0 of cad-viewer-widget being registered

[image]

In Jupyter I see

[image]

Do you see the same?

What does the browser console error show. It should show a version of the model it expects?

altaic, 2022-02-15:
@bernhard-42 The `jupyter-labextension list` output is the same, as is the version output from the print statements in the jupyter notebook. However, the console log shows:
```
523.cf94b68f6edde7eefd61.js:1 Uncaught (in promise) Error: Module cad-viewer-widget, semver range 1.0.0 is not registered as a widget module
    at x.loadClass (523.cf94b68f6edde7eefd61.js:1)
    at x.<anonymous> (272.9d70a85a9abe209402d0.js:1)
    at 272.9d70a85a9abe209402d0.js:1
    at Object.next (272.9d70a85a9abe209402d0.js:1)
    at 272.9d70a85a9abe209402d0.js:1
    at new Promise (<anonymous>)
    at S (272.9d70a85a9abe209402d0.js:1)
    at x.e._make_model (272.9d70a85a9abe209402d0.js:1)
    at x.<anonymous> (272.9d70a85a9abe209402d0.js:1)
    at 272.9d70a85a9abe209402d0.js:1
    at Object.next (272.9d70a85a9abe209402d0.js:1)
    at 272.9d70a85a9abe209402d0.js:1
    at new Promise (<anonymous>)
    at S (272.9d70a85a9abe209402d0.js:1)
    at x.e.new_model (272.9d70a85a9abe209402d0.js:1)
    at x.e.handle_comm_open (272.9d70a85a9abe209402d0.js:1)
loadClass @ 523.cf94b68f6edde7eefd61.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
S @ 272.9d70a85a9abe209402d0.js:1
e._make_model @ 272.9d70a85a9abe209402d0.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
S @ 272.9d70a85a9abe209402d0.js:1
e.new_model @ 272.9d70a85a9abe209402d0.js:1
e.handle_comm_open @ 272.9d70a85a9abe209402d0.js:1
_handleCommOpen @ 523.cf94b68f6edde7eefd61.js:1
_handleCommOpen @ default.js:1038
Promise.then (async)
e.register_model @ 272.9d70a85a9abe209402d0.js:1
register_model @ 523.cf94b68f6edde7eefd61.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
(anonymous) @ 272.9d70a85a9abe209402d0.js:1
S @ 272.9d70a85a9abe209402d0.js:1
e.new_model @ 272.9d70a85a9abe209402d0.js:1
e.handle_comm_open @ 272.9d70a85a9abe209402d0.js:1
_handleCommOpen @ 523.cf94b68f6edde7eefd61.js:1
_handleCommOpen @ default.js:1038
async function (async)
_handleCommOpen @ default.js:1037
_handleMessage @ default.js:1206
async function (async)
_handleMessage @ default.js:1168
(anonymous) @ default.js:110
Promise.then (async)
KernelConnection._onWSMessage @ default.js:107
```

```
523.cf94b68f6edde7eefd61.js:1 Uncaught (in promise) Error: Module cad-viewer-widget, semver range 1.0.0 is not registered as a widget module
    at x.loadClass (523.cf94b68f6edde7eefd61.js:1)
    at x.<anonymous> (272.9d70a85a9abe209402d0.js:1)
    at 272.9d70a85a9abe209402d0.js:1
    at Object.next (272.9d70a85a9abe209402d0.js:1)
    at 272.9d70a85a9abe209402d0.js:1
    at new Promise (<anonymous>)
    at S (272.9d70a85a9abe209402d0.js:1)
    at x.e._make_model (272.9d70a85a9abe209402d0.js:1)
    at x.<anonymous> (272.9d70a85a9abe209402d0.js:1)
    at 272.9d70a85a9abe209402d0.js:1
    at Object.next (272.9d70a85a9abe209402d0.js:1)
[... 92 lines cut ...]
async function (async)
_handleCommOpen @ default.js:1037
_handleMessage @ default.js:1206
async function (async)
_handleMessage @ default.js:1168
(anonymous) @ default.js:110
Promise.then (async)
KernelConnection._onWSMessage @ default.js:107
16:26:23.243 › Exception opening new comm
setTimeout (async)
consoleLog @ console.js:93
transport @ console.js:46
runTransport @ log.js:44
runTransports @ log.js:27
log @ log.js:21
_handleCommOpen @ default.js:1044
async function (async)
_handleCommOpen @ default.js:1037
_handleMessage @ default.js:1206
async function (async)
_handleMessage @ default.js:1168
(anonymous) @ default.js:110
Promise.then (async)
KernelConnection._onWSMessage @ default.js:107
```

bernhard-42, 2022-02-15:
I just saw 

> I have nb_conda_kernels installed in the base environment, which makes all conda envs available as jupyterlab kernels.

Does that mean you have both rc2 and rc3 installed and you are just switching kernels in Jupyterlab?
I just tried it and executed 

```
%%sh
jupyter-labextension list
```

in a notebook cell. As expected, switching kernel does not change the extensions.

If rc2 works, could it be that you started jupyterlab from the rc2 environment and then try to switch kernel?

bernhard-42, 2022-02-15:
Does the error also appear when you start Jupyterlab in the py39-jcqrc3 environment and use the "Python3 (ipykernel)" kernel? 

btw, when `cad-viewer-widget: Widget version 1.0.0 is registered` doesn't appear in the browser console when you open Jupyterlab (without any cell executed), then `cad-viewer-widget` is not properly installed or configured. 

This can happen when you you have an older version of `cad-viewer-widget` installed, run some CadQuery stuff and then update `cad-viewer-widget` but only restart the kernel in Jupyterlab. This is not sufficient. One needs to also refresh the browser (F5, cmd-R) to ensure Jupyterlab loads the newest Javascript lib that fits to the new Python part.

altaic, 2022-02-15:
> Looks like the issue is introduced in [bernhard-42/cad-viewer-widget@f1d03ec](https://github.com/bernhard-42/cad-viewer-widget/commit/f1d03ecb0cb5779d8b725f83fbd4a531eb32dc04). The previous commit works fine.

To expand on this comment, I set up two conda envs:
```
name: cvw-e6b530
channels:
  - defaults
  - cadquery
  - conda-forge
dependencies:
  - python=3.9
  - pip
  - cadquery=master
  - matplotlib=3.3
  - pip:
    - git+https://github.com/bernhard-42/cad-viewer-widget.git@e6b530ea9861e99744ea8c0e73dcb89b8ef75658  # commit: "adapted the examples"
```
and
```
name: cvw-f1d03e
channels:
  - defaults
  - cadquery
  - conda-forge
dependencies:
  - python=3.9
  - pip
  - cadquery=master
  - matplotlib=3.3
  - pip:
    - git+https://github.com/bernhard-42/cad-viewer-widget.git@f1d03ecb0cb5779d8b725f83fbd4a531eb32dc04  # commit: "latest release 1.0.0"
```

Then I ran the following in the jupyter notebook with each of the above kernels:
```
from cad_viewer_widget import open_viewer
open_viewer()
```

The `cvw-e6b530` kernel displayed the viewer (albeit all white), while the `cvw-f1d03e` kernel displayed `Error displaying widget: model not found` along with the same `Uncaught (in promise) Error: Module cad-viewer-widget, semver range 1.0.0 is not registered as a widget module` in the console.

The last commit in `cad-viewer-widget` entitled "Latest release: 1.0.0" appears to be the culprit. It looks like the only thing that really changed there is the switch to `three-cad-viewer` version 1.3.1 from version 1.2.8.

altaic, 2022-02-15:
> Does the error also appear when you start Jupyterlab in the py39-jcqrc3 environment and use the "Python3 (ipykernel)" kernel?
>
> btw, when `cad-viewer-widget: Widget version 1.0.0 is registered` [...]

Ah, that must be it. I must have installed jupyter-cadquery in the base conda environment at some point... Yes, indeed, there's a bunch of junk in there.

altaic, 2022-02-15:
> I just saw
>
>
> Does that mean you have both rc2 and rc3 installed and you are just switching kernels in Jupyterlab? I just tried it and executed
>
> ```
> %%sh
> jupyter-labextension list
> ```
>
> in a notebook [...]

In an attempt to clean out the cruft from the base env, I've managed to break my jupyterlab install. I'll reinstall and see if that fixes things for the time being. Sorry for the wild goose chase!

Regardless of the clean install fix, though, it seems I've made some poor assumptions about kernels in jupyterlab. I thought that I could have a few different conda environments and use them as kernels in jupyterlab. Is there a better way to accomplish that than nb_conda_kernels?

altaic, 2022-02-16:
I tried a fresh `jupyterlab-desktop` install with the `py39-jcqrc3` environment specs above, and I'm still getting the same error.

bernhard-42, 2022-02-16:
It looks like you are using [jupyterlab-desktop](https://github.com/jupyterlab/jupyterlab-desktop) on MacOS, right?
I need to to test Jupyter-CadQuery with it and first install rc2 and then rc3.

The tricky thing about extensions is that they have two parts:
1) the python part running in the python interpreter of the selected kernel
2) the javascript part for the browser frontend of Jupyterlab
And both need to have the same version in order to work correctly

Maybe in jupyterlab-desktop frontend config gets handled differently, but I need to test.

Have you ever tried the "normal" JupyterLab way, i.e.

```shell
conda create -n jcq3 -c conda-forge -c cadquery python=3.8 cadquery=master
conda activate jcq3
pip install jupyter-cadquery==3.0.0rc3 matplotlib
jupyter lab
```
?

bernhard-42, 2022-02-16:
I've now installed two conda envs with your yaml files given above
I can run both in parallel and it works

I even used the same notebook at the same time 
[image]

What you can see is the model error in the left window. It happens when I run the code in the right window (now rc2 is stored in the output of the notebook) and then refresh the left side. Since it has rc3 installed it can show the rc2 model of the output created by rc2. But when I execute the code in the left window, all works and now rc3 is in the output of the notebook  (btw. don't use one notebook from two Jupyterlabs - huge mess will happen)

I will now give jupyterlab-desktop a try

bernhard-42, 2022-02-16:
I have installed `jupyterlab-desktop` and used the kernel switcher as described here: https://github.com/jupyterlab/jupyterlab-desktop/blob/master/user-guide.md#changing-the-python-environment-used-by-jupyterlab-desktop

I switched to kernel rc2

[image]

and ran the example (look at the versions)

[image]

I then switched to kernel rc3

[image]

and ran the example (look at the versions)

[image]

**Note**: 
- `jupyterlab-desktop` does a full restart after switching the environment. By that it ensures to load the correct javascript part for the environment.
- I did not use [nb_conda_kernels](https://github.com/Anaconda-Platform/nb_conda_kernels), since it doesn't do a restart of `jupyterlab-desktop`

altaic, 2022-02-17:
> **Note**:
>
> * `jupyterlab-desktop` does a full restart after switching the environment. By that it ensures to load the correct javascript part for the environment.
> * I did not use [...]

Ah, that's the key. Thank you for helping me figure this out. I also found [this comment](https://github.com/Anaconda-Platform/nb_conda_kernels/issues/201#issuecomment-901068937) quite helpful for understanding ipywidgets.

So, essentially, to use a given widget, it must be installed in whatever environment jupyterlab was started in, and all kernels that use the widget must have the same version installed. In jupyterlab-desktop, you can switch the environment (rather than switching kernels), though this also restarts the app. It's all rather awkward, but it seems that's just how jupyterlab is designed.

closed 2022-02-17 by altaic

bernhard-42, 2022-02-17:
> It's all rather awkward, but it seems that's just how jupyterlab is designed.

yes, but the benefit is that you can now interact between the model in python and the view in Javascript, see the examples, especially the first in my repo

chemiskyy, 2022-03-21:
I run into the same issue as Altaic using Safari for Jupiter lab. I am not using nb_conda_kernels. In Firefox everything is working fine