bernhard-42/jupyter-cadquery #64: Installation on ipad using carnets

usdk1, 2022-02-16:
Hi,
I'm relatively new to pyhton and cadquery, so hopefully this is the right place to ask....
cadquery is the tool of joice to implement program based designs, the ipad my favorite tool for new design ideas.

With the Carnets app (https://github.com/holzschu/Carnets) I can get a Jupyter Notebook on my Ipad pro. I could install Version 3.0.0rc3 of jupyter-cadquery.
But: Carpets does not support conda (only pip) and I don't know how get Jupyterlab up and running in this (or an other?) environment on my ipad. Any idea how to?

bernhard-42, 2022-02-16:
Hi @usdk1,
I am sorry to say, this will not work. Even if we would somehow get `Jupyter-Cadquery` installed (both the Python module and the Javascript extension), the underlying CAD environment based on [CadQuery](https://github.com/CadQuery/cadquery) will not work. CadQuery depends on a C++ based CAD kernel (OpenCascade wrapped with [OCP](https://github.com/CadQuery/OCP)) and there is no way to get this running on an iPad.

usdk1, 2022-02-17:
Good Morning Bernhard,
Thanks for your support and explanation. Some more comments on OCP, OCCT and opencascade:
I just installed CAD Assistant https://dev.opencascade.org/project/cad-assistant on my IPAD and it seems to be based on OpenCascade. Sharpr seems also to be based on OCC. So in principle it seems there is a path. No glue whether this has any meaning for you?

I also got an answer from the developer of the Carnets app, just to share the information… https://github.com/holzschu/Carnets/issues/231#issuecomment-1041499588

bernhard-42, 2022-02-17:
Thanks for linking to Carnet, please see my answer there to better understand the opencascade issue. Of course, I can build a native iPadOS app with opencascade, but the python module approach is not compatible with iPadOS afaik

closed 2022-03-13 by bernhard-42