bernhard-42/jupyter-cadquery #65: Docker image for 3.0.0rc3 crashes on Mac M1

bracki, 2022-02-17:
```
docker run -it --rm  -p 8888:8888 bwalter42/jupyter_cadquery:3.0.0rc3

WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
Starting in JupyterLab mode: http://localhost:8888/lab
[W 2022-02-17 09:47:22.274 LabApp] 'token' has moved from NotebookApp to ServerApp. This config will be passed to ServerApp. Be sure to update your config before our next release.
[W 2022-02-17 09:47:22.276 LabApp] 'allow_origin' has moved from NotebookApp to ServerApp. This config will be passed to ServerApp. Be sure to update your config before our next release.
[I 2022-02-17 09:47:22.314 ServerApp] jupyterlab | extension was successfully linked.
[I 2022-02-17 09:47:22.368 ServerApp] Writing Jupyter server cookie secret to /home/cq/.local/share/jupyter/runtime/jupyter_cookie_secret
[I 2022-02-17 09:47:24.890 ServerApp] nbclassic | extension was successfully linked.
[I 2022-02-17 09:47:24.891 ServerApp] voila.server_extension | extension was successfully linked.
[W 2022-02-17 09:47:25.118 ServerApp] All authentication is disabled.  Anyone who can connect to this server will be able to run code.
[I 2022-02-17 09:47:25.143 ServerApp] nbclassic | extension was successfully loaded.
[I 2022-02-17 09:47:25.150 LabApp] JupyterLab extension loaded from /opt/conda/envs/cq/lib/python3.8/site-packages/jupyterlab
[I 2022-02-17 09:47:25.150 LabApp] JupyterLab application directory is /opt/conda/envs/cq/share/jupyter/lab
[I 2022-02-17 09:47:25.169 ServerApp] jupyterlab | extension was successfully loaded.
[I 2022-02-17 09:47:25.180 ServerApp] voila.server_extension | extension was successfully loaded.
[I 2022-02-17 09:47:25.182 ServerApp] Serving notebooks from local directory: /home/cq
[I 2022-02-17 09:47:25.182 ServerApp] Jupyter Server 1.13.5 is running at:
[I 2022-02-17 09:47:25.183 ServerApp] http://51d9fbc2f560:8888/lab
[I 2022-02-17 09:47:25.183 ServerApp]  or http://127.0.0.1:8888/lab
[I 2022-02-17 09:47:25.184 ServerApp] Use Control-C to stop this server and shut down all kernels (twice to skip confirmation).
[W 2022-02-17 09:47:31.342 LabApp] Could not determine jupyterlab build status without nodejs
[I 2022-02-17 09:47:39.776 ServerApp] Writing notebook-signing key to /home/cq/.local/share/jupyter/notebook_secret
[W 2022-02-17 09:47:39.784 ServerApp] Notebook 1-cadquery.ipynb is not trusted
Operation not permitted (src/thread.cpp:309)
Operation not permitted (src/thread.cpp:309)
qemu: uncaught target signal 6 (Aborted) - core dumped
/tmp/run.sh: line 39:    27 Aborted                 jupyter lab --ip=0.0.0.0 --no-browser --NotebookApp.token='' --NotebookApp.allow_origin='*'
```

Jupyter launches. When I click on one of the notebook it says "Error starting kernel, load failed:.

bernhard-42, 2022-02-17:
@bracki What Docker app are you running on the M1?
I have decided for [colima](https://github.com/abiosoft/colima), which is a great replacement for the Docker Desktop (which I left due to their changed terms, and it works well)

```bash
$ uname -msr
Darwin 21.2.0 arm64
```

```bash
$ docker run -it --rm  -p 8888:8888 bwalter42/jupyter_cadquery:3.0.0rc3
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
Starting in JupyterLab mode: http://localhost:8888/lab
[W 2022-02-17 12:34:15.874 LabApp] 'token' has moved from NotebookApp to ServerApp. This config will be passed to ServerApp. Be sure to update your config before our next release.
[W 2022-02-17 12:34:15.876 LabApp] 'allow_origin' has moved from NotebookApp to ServerApp. This config will be passed to ServerApp. Be sure to update your config before our next release.
[I 2022-02-17 12:34:15.913 ServerApp] jupyterlab | extension was successfully linked.
[I 2022-02-17 12:34:15.964 ServerApp] Writing Jupyter server cookie secret to /home/cq/.local/share/jupyter/runtime/jupyter_cookie_secret
[I 2022-02-17 12:34:18.383 ServerApp] nbclassic | extension was successfully linked.
[I 2022-02-17 12:34:18.384 ServerApp] voila.server_extension | extension was successfully linked.
[W 2022-02-17 12:34:18.601 ServerApp] All authentication is disabled.  Anyone who can connect to this server will be able to run code.
[I 2022-02-17 12:34:18.624 ServerApp] nbclassic | extension was successfully loaded.
[I 2022-02-17 12:34:18.629 LabApp] JupyterLab extension loaded from /opt/conda/envs/cq/lib/python3.8/site-packages/jupyterlab
[I 2022-02-17 12:34:18.629 LabApp] JupyterLab application directory is /opt/conda/envs/cq/share/jupyter/lab
[I 2022-02-17 12:34:18.644 ServerApp] jupyterlab | extension was successfully loaded.
[I 2022-02-17 12:34:18.653 ServerApp] voila.server_extension | extension was successfully loaded.
[I 2022-02-17 12:34:18.655 ServerApp] Serving notebooks from local directory: /home/cq
[I 2022-02-17 12:34:18.656 ServerApp] Jupyter Server 1.13.5 is running at:
[I 2022-02-17 12:34:18.656 ServerApp] http://3fa4fc36fe8e:8888/lab
[I 2022-02-17 12:34:18.656 ServerApp]  or http://127.0.0.1:8888/lab
[I 2022-02-17 12:34:18.656 ServerApp] Use Control-C to stop this server and shut down all kernels (twice to skip confirmation).
[W 2022-02-17 12:34:42.390 LabApp] Could not determine jupyterlab build status without nodejs
[I 2022-02-17 12:34:46.748 ServerApp] Writing notebook-signing key to /home/cq/.local/share/jupyter/notebook_secret
[W 2022-02-17 12:34:46.756 ServerApp] Notebook 1-cadquery.ipynb is not trusted
[I 2022-02-17 12:34:47.518 ServerApp] Kernel started: 17ebc110-9cad-4591-926c-5ca94d2877e2
[W 2022-02-17 12:34:54.840 ServerApp] Got events for closed stream <zmq.eventloop.zmqstream.ZMQStream object at 0x4019a03610>
```

[image]

See hostname in Jupyter output and the "Jupyter Server 1.13.5 is running at" address in the log above. 

So under `colima` all runs.

bernhard-42, 2022-08-02:
Since it runs under `colima` and I can't add anything to docker on Mac being stable, I will close this