bernhard-42/jupyter-cadquery #66: Consider adding to Jupyter Docker Stacks

okjed, 2022-03-12:
We learned about your project in a great Open Source Hardware meeting today and we were amazed about the possibilities. I was curious if you'd considered contributing to the Jupyter Docker Stacks to make it easy for those of us that deploy JupyterHub on Kubernetes to make it much simpler to use your project.

https://jupyter-docker-stacks.readthedocs.io/en/latest/index.html

I may look more into what would need to occur to build a specific stack but felt this would be great for easy of operations for many of us that already leverage the environment.

bernhard-42, 2022-03-12:
@okjed I already have a docker version ([Dockerfile](https://github.com/bernhard-42/jupyter-cadquery/blob/master/docker/Dockerfile ))
Usage: `docker run -it --rm -v $WORKDIR:/home/cq -p 8888:8888 bwalter42/jupyter_cadquery:3.0.0` 
For more details, see https://github.com/bernhard-42/jupyter-cadquery#using-a-docker-image 
And it runs in Binder.
Maybe that helps?

okjed, 2022-03-12:
Yes, that is what encouraged me to open the issue and how we did the demos during the meeting today. A lot of universities are deploying JupyterHub on bare metal kubernetes and it eases the ability of picking a specific and appropriate Jupyter stack for projects. I'll look into converting the existing Dockerfile to match the criteria once I am more comfortable with both projects. To date, I've only augmented existing Jupyter Docker Stacks - here is the link I should've shared earlier: https://github.com/jupyter/docker-stacks

bernhard-42, 2022-03-12:
@okjed Here is a minimal version as a starting point. It is built according to https://github.com/jupyter/docker-stacks/blob/master/scipy-notebook/Dockerfile

```Dockerfile
# Copyright (c) Jupyter Development Team.
# Distributed under the terms of the Modified BSD License.
ARG OWNER=jupyter
ARG BASE_CONTAINER=$OWNER/minimal-notebook
FROM $BASE_CONTAINER

LABEL maintainer="Jupyter Project <jupyter@googlegroups.com>"

USER root

RUN apt-get update -y && \
    apt-get install --no-install-recommends -y libgl1-mesa-glx libglu1-mesa && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

USER ${NB_UID}

RUN mamba install -y -c default -c conda-forge -c cadquery python=3.9 cadquery=master && \
    mamba clean --all -f -y && \
    pip install jupyter-cadquery==3.0.0 cadquery-massembly~=0.9 && \
    fix-permissions "${CONDA_DIR}" && \
    fix-permissions "/home/${NB_USER}"

USER ${NB_UID}

WORKDIR "${HOME}"
```

Then build it

```bash
docker build -t jupyter/cadquery .
```

and run it:

```bash
docker run -p 10000:8888 jupyter/cadquery
```

No idea how the docker-stacks folks deal with the access token of jupyterlab, but if you copy and paste it, this is the result

[image]

okjed, 2022-03-13:
Thank you very much! This worked perfectly and allowed me to add it to a scaled up JupyterHub instance that we use in a class environment! Incredible project!

closed 2022-03-13 by okjed

bernhard-42, 2022-03-13:
I only do the viewer in Jupyterlab. 

The whole CAD library ([CadQuery](https://github.com/CadQuery/cadquery)) is done by @adam-urbanczyk on top of [Open Cascade](https://www.opencascade.com/)