bernhard-42/jupyter-cadquery #67: How to use the standalone viewer with Docker?

roipoussiere, 2022-03-23:
The readme explains how to run the standalone viewer with Docker:

>    docker run -it --rm -p 8888:8888 -p 5555:5555 bwalter42/jupyter_cadquery:3.0.0 -v [-d]

This displays the cadquery logo, but I don't understand how to run a specific python script located on the host.

roipoussiere, 2022-03-23:
I just read [this comment](https://github.com/bernhard-42/jupyter-cadquery/issues/55#issuecomment-934473639) suggesting that it can be used with VS Code using the "Remote - Containers" extension.

I don't use VSCode (I use Codium which does not include this extension), so I can't do this.

So I wonder if the standalone viewer can be used among an external IDE with a more generic way.

For instance by running the viewer in a docker container and attach to it a Python script to a docker volume, then each time the file is modified on the host (= saved on the IDE), the viewer is updated.

bernhard-42, 2022-03-24:
@roipoussiere I was able to it via `docker exec -i jcq bash -c ". /opt/conda/bin/activate cq; python" < t.py`

roipoussiere, 2022-03-24:
Thank you, it works!

More precisely, I needed to run the command twice in order to make it work: after the first command I get the error ` ValueError: /bottom/right_back is not a valid subpath of any of ['/jupyter cadquery/jupyter', '/jupyter cadquery/cadquery']`, which disapear the second time. I don't really understand why.

Also, can I make a PR to add this useful command on the readme under the section *Using a docker image / standalone viewer*?

bernhard-42, 2022-03-24:
`/bottom/right_back` is a path to a part of the sample hexapod assembly. `['/jupyter cadquery/jupyter', '/jupyter cadquery/cadquery']` are objects of the logo assembly. Totally different views, strange ...

I tried to start the docker container and issued 

`docker exec -i 333d01a642ec  bash -c ". /opt/conda/bin/activate cq; python" < 2-hexapod.py` 

and had no issue (333d01a642ec is my jcq container). 

Can you reproduce it?

And yes, happy to accept a PR.

roipoussiere, 2022-03-24:
Hmm, can not reproduce this exact error, but I still need no run the script twice:

```
docker stop jcq
jcq
❯ docker run -d --rm -p 8888:8888 --name jcq bwalter42/jupyter_cadquery:3.0.0 -v -d
f8d42f5041e590a82bc755c554cd5894cbbbf06aa542911bb62aa602e916444d
❯ docker exec -i jcq bash -c ". /opt/conda/bin/activate cq; python" < ./2-hexapod.py
.............................................. sending ... Reconnecting to server…
Resending ...
Reconnecting to server…
Resending ...
 sending ... Reconnecting to server…
Resending ...
Reconnecting to server…
Resending ...
# when I go to 127.0.0.1:8888, the cadqery logo is displayed instead the hexapod
❯ docker exec -i jcq bash -c ". /opt/conda/bin/activate cq; python" < ./2-hexapod.py
.............................................. sending ... done
 sending ... done
# and from here the hexapod is displayed correctly
```

closed 2022-03-25 by roipoussiere