bernhard-42/jupyter-cadquery #68: VSCode Extension

bernhard-42, 2022-03-24:
Now, with Jupyter-CadQuery 3 the actual CAD viewer is full written in Javascript [three-cad-viewer](https://github.com/bernhard-42/three-cad-viewer). So it should be possible to write a (native) VS Code extension: 

One would use python code from this repo. For assemblies this would be

```python
from jupyter_cadquery.utils import numpy_to_json
from jupyter_cadquery.cad_objects import to_assembly
from jupyter_cadquery.base import _tessellate_group

result = numpy_to_json(_tessellate_group(to_assembly(a)))
```
See https://github.com/bernhard-42/jupyter-cadquery/blob/7ed5ff2ba484ea6cb96f5ea999a88484dfb6b1c1/jupyter_cadquery/cad_objects.py#L616-L622 and https://github.com/bernhard-42/jupyter-cadquery/blob/7ed5ff2ba484ea6cb96f5ea999a88484dfb6b1c1/jupyter_cadquery/base.py#L482 

The output is the format of the three-cad-viewer component. So one needs to run the python code above from VS Code and then push the resulting json object into three-cad-viewer which runs in a web view of VS Code.

However, I don't have time to maintain another project. But I am happy to help if someone gives it a try ;-)

roipoussiere, 2022-03-28:
I promise nothing, but I'm going to give a try. :)

roipoussiere, 2022-03-29:
Tadaaa!

[image]

Ok it's not functional yet (the rendering is not generated by the code dynamically), but it's a work in progress.

I will publish the extension when it will be functional (soon I hope). :)

bernhard-42, 2022-03-31:
@roipoussiere This is awesome! I really like the idea of the cadquery server. Great user experience

roipoussiere, 2022-03-31:
Thank you! I'm pretty happy because it's my first VSCode extension and I managed to get a prototype in 2-3 days. :)

Concerning the cadquery server, a user on Discord pointed me that a non-docker version (that work with a local cadquery installation) would be useful for users that work on a custom version on cadquery, so I think it will be my next step.

bernhard-42, 2022-04-01:
Additionally, please think about how to enable debugging.
In an IDE this is key (together with auto-completion). I haven't found a way to use debugging, since you send the code to the server

roipoussiere, 2022-04-01:
What do you want to mean mean exactly by debugging?

I can easily provide the stdin and stdout, but do you think about something else?

bernhard-42, 2022-04-01:
Use the VS Code python debugger and step through the code, looking at variables in each step to find issues. For larger models e.g. organized in classes, this would be really helpful.
For me that would be the only reason to move from Jupyter to an IDE

bernhard-42, 2022-04-01:
This is something my Viewer approach allows. The Viewer approach is not very convenient, but powerful. I can have several show steps to even visually check my model during step-by-step debugging

roipoussiere, 2022-04-01:
I see. It will not be trivial but possible.

I could update the cadquery server api to provide a debug mode: on this mode, the code is send line by line to the server, among with a token to identify the script execution. On the server, I use python introspection to retrieve the value of the current variables, then I send them back to VSCode, on which I integrate them into the VSCode debugger.

roipoussiere, 2022-04-01:
Hmm, in fact VSCode provide a way to use [a remote Python interpreter](https://devblogs.microsoft.com/python/remote-python-development-in-visual-studio-code/). I will investigate this.

bernhard-42, 2022-06-19:
A solution based on ˋjupyter-cadqueryˋ is avaiable at https://framagit.org/roipoussiere/cadquery-vscode
So closing this issue

closed 2022-06-19 by bernhard-42