bernhard-42/jupyter-cadquery #69: transparent sidebar backgound

roipoussiere, 2022-03-24:
A such option would be appreciated to save space, which is necessary when working with an IDE.

Inspirations:

FreeCAD with [Glass addon](https://github.com/triplus/Glass):
[image]

SolidWorks with transparent FeatureManager:
[image]

bernhard-42, 2022-03-26:
@roipoussiere I've built a new version that features a glass mode ;-) 

Would you like to give it a try and provide feedback? https://bernhard-42.github.io/three-cad-viewer/example.html

Btw. it also features auto-switch to dark mode if browser is used in dark mode

[image]

bernhard-42, 2022-03-26:
The minimum width needed is `cadWidth=800`
Below 1000, "Transparent", "Black Edges" and "Explode" will be abbreviated. For `cadWidth >= 1000` they are kept in long form.

I don't want to change the UI too much in this major release.  Maybe I group the radio buttons behind some drop down menu later, ...

bernhard-42, 2022-03-26:
Abbreviations are ugly ...
Changed the menu slightly

[image]

Now the minimum width is 700px for glass mode.

What do you think?

roipoussiere, 2022-03-28:
> I've built a new version that features a glass mode ;-)

Awesome! 😍

There is a thin border around the tree area in darkmode, that doesn't appear on you last screenshot, is it intentional?

[image]

> Btw. it also features auto-switch to dark mode if browser is used in dark mode

Perfect! 😍

> Changed the menu slightly

I like it! 😍

bernhard-42, 2022-03-28:
> There is a thin border around the tree area in darkmode

What OS and browser are you using?

Brave  / Mac
[image]

Firefox / Mac
[image]

So on the Mac

roipoussiere, 2022-03-28:
Ubuntu 21.10, Firefox 98

roipoussiere, 2022-03-28:
The discussion can continue on #77.

Thank you so much for your reactivity. :)

closed 2022-03-28 by roipoussiere