bernhard-42/jupyter-cadquery #70: radio button icons

roipoussiere, 2022-03-24:
I would like to try some UI improvements by using icons on radio buttons instead of text.

Here a first try with the `axis` button:
[image]

The icon itself is in vector format and created with Inkscape with the same stroke style.
The background would change to dark to light to inform button state.
The html/css is not implemented yet, I realized this picture on Gimp.

What do you think about this, should I continue for the other buttons?

roipoussiere, 2022-03-24:
> All Icons have been hand crafted in the notebook

Understood.

> Overall I am not sure this would make the UI look better

I personally appreciate icons, usually. A more objective benefit here is that it save space so the top bar could fit in a half-screen in order to work with an IDE.

> what would be an icon for "transparent" or "orth"

Definitely a challenge here, but probably not impossible.

> would people understand the icons?

They can still use the tooltip! :)

Also, on some software it's possible to switch buttons style from text, icon or both.

In fact, my main motivation was a bad user experience about these buttons. If you don't want icons I would at least suggest to:
- move the radio button on the left (it seem more intuitive for me, I think many software and websites does this);
- make the text clickable as well.

bernhard-42, 2022-03-24:
> - move the radio button on the left (it seem more intuitive for me, I think many software and websites does this);
> - make the text clickable as well.

Agreed, this is a long topic to resolve, but due to all the challenges in the latest release, I again forgot it. I just opened a ticket for it.

What I currently like about radio buttons being text and buttons being icons is that the UI element categories look clearly different. But of course, this is just my pov

bernhard-42, 2022-03-24:
My development version already works as expected. So the next release will have it.
[image]

roipoussiere, 2022-03-25:
Much better! I'm closing this now.

closed 2022-03-25 by roipoussiere