bernhard-42/jupyter-cadquery #72: Hexapod mates example: axes don't follow objects when using explode function

roipoussiere, 2022-03-25:
Steps to reproduce: 
1. Got to https://bernhard-42.github.io/three-cad-viewer/example.html;
2. check **explode** option;
3. click on play: the axes doesn't follow the trajectory of their related objects.

[image]

I don't know if this is an error on the example or on the jcq project...

bernhard-42, 2022-03-25:
Still fine tuning ...

bernhard-42, 2022-03-25:
The explode algorithm calculates the center of each object (not of the group including the underlying hierarchy)
It will then use as a direction for each object the axis from the origin (or the bounding box center) to the center of the object.
As such a mate and the object it belongs to usually will have a different angle of the trajectory (since center of mate and center of its object are not identical)

bernhard-42, 2022-03-31:
@roipoussiere This should now be fixed with 1.5.6
[image]

roipoussiere, 2022-03-31:
In the current live example, I get this:

[image]

When I click on explode, I was expecting the small axes to be positioned on the corresponding axes on the parts.

But I suppose the axes themselves are concerned by the *explode*, so they can't be aligned anymore, so it's probably a correct behavior.

bernhard-42, 2022-04-01:
Got it. 
For the explode algorithm, the mates/axes are simple objects without meaning. Hence they will be exploded properly. One advantage is that one can see alinement mates and how they align by going backwards in the explosion animation.
In general, helpers like mates are not the best use case for the explode algorithm

roipoussiere, 2022-04-01:
Ok I think this can be closed now, since you fixed this.

closed 2022-04-01 by roipoussiere