bernhard-42/jupyter-cadquery #74: Highlight the selected view

roipoussiere, 2022-03-25:
When the user selects a view (like iso, front, etc), I could be practical to highlight it until the view is updated.

This way the user can quickly see which one is used.

bernhard-42, 2022-03-25:
You mean keep the button highlighted until another button is pressed or the object is moved?

roipoussiere, 2022-03-25:
both

bernhard-42, 2022-03-25:
Pretty easy to implement and actually helpful:

[image]

Is this what you had in mind?

Of course, if one rotates the object and hits by chance one of the predefined directions, buttons will not be highlighted

roipoussiere, 2022-03-25:
> Is this what you had in mind?

Yep, exactly!

bernhard-42, 2022-03-26:
I'd like to keep it subtle ...

roipoussiere, 2022-03-28:
Oh I just realized that this was actually published on https://github.com/bernhard-42/three-cad-viewer/commit/4cfe20439b8975c2dd5cf8ccab028e859fe86370, so I'm closing this now, thank you!

closed 2022-03-28 by roipoussiere