bernhard-42/jupyter-cadquery #75: Move the animation bar on the top

roipoussiere, 2022-03-28:
The animation bar already appears trunked on my (1680*1050) monitor:

[image]

You can notice the scroll bar on the right, meaning the content doesn't fit on the screen.

It's annoying because I must:
- scroll up each time I want to access the top tool bar;
- scroll down each time I want to access to animation bar.

I know that it's harder to correctly position elements vertically than horizontally, so I would suggest to move the bar upper on the screen, for instance on the right of the tree panel:

[image]

bernhard-42, 2022-03-28:
`jcv` has two parameters `-h` and `-w`. Unfortunately, they are not supported by the docker version. I need to add them. 
So the right approach will be to set the height when starting jcv. I've created https://github.com/bernhard-42/jupyter-cadquery/issues/76

bernhard-42, 2022-03-28:
@roipoussiere Please try
`docker run -it --rm -p 8888:8888 --name jcq -e "CAD_HEIGHT=520" -e "CAD_WIDTH=700" -p 5555:5555 bwalter42/jupyter_cadquery:3.0.0 -v`
This should open a small window and the animation bar should be visible.

closed 2022-03-28 by roipoussiere