bernhard-42/jupyter-cadquery #78: add options to hide some tools in toolbar

roipoussiere, 2022-03-29:
When using the viewer with a small width, the toolbar might be too wide.

Hiding some tools will make it lighter, but in order to still be able to use them if necessary, I would suggest 3 states:
- **hidden**: the tool is not visible on the bar;
- **visible**: the tool is not visible on the bar;
- **collapsed**: the tool is moved in the *More* dropdown list;

I'm not sure what is the best way to do it. Maybe adding two more options that use tools identifiers in arrays:

```js
const options = {
    // ...
    hide_tools: ['ortho', 'explode', 'axes0', 'pin_view'],
    collapse_tools: ['axes', 'grid', 'transparent', 'black_edge', 'explode', 'help']
}
```

bernhard-42, 2022-03-29:
I'm not sure I want to spend time on this. We are now down to 700px width of the CAD window which is an overall size of 700x525. I don't believe working with CAD models on smaller canvases really makes sense

What is your use case that 700px width is too large?

roipoussiere, 2022-03-29:
> What is your use case that 700px width is too large?

The VSCode extension I'm working on:

[image]

I don't want to impose a large view panel to users, neither to force them to hide the VSCode tree view (that I plan to use later to replace the CadQuery tree view).

The options I suggested above will be defined in the extension option by the user according to their needs.

bernhard-42, 2022-03-29:
What width do you want to achieve? Just to understand whether it would be achievable...

roipoussiere, 2022-03-29:
Ideally a minimum of 450px, which is just enough to display the view buttons:

[image]

bernhard-42, 2022-03-29:
You could also use the "no tools" mode and add your own buttons? You then have the full freedom of width and height (and I would just remove the limitation of width>=700px)

This is where you would start:

https://github.com/bernhard-42/three-cad-viewer/blob/ea80d773ddabc60b454d8f886d217833cdbf199d/src/display.js#L511-L525

roipoussiere, 2022-03-29:
Ok I think I can do that, thanks!

bernhard-42, 2022-03-29:
but you wouldn't have the tree, comes to my mind now.
How about this:
You use the glass mode with width 450px leading to

[image]

Then run this code in the browser console:

```javascript
tags = ["tcv_axes", "tcv_axes0", "tcv_grid", "tcv_ortho"]
tags.forEach((tag) => {
    console.log(1, tag)
    var el = viewer.display._getElement(tag);
    console.log(2, el)
    el.style.display = "none";
    el.nextElementSibling.style.display = "none";
})
```

It should lead to 

[image]

Would that make sense?

bernhard-42, 2022-03-29:
and of course, you would remove the pin button in vscode.

[image]

roipoussiere, 2022-03-29:
Thank you! I will investigate this later.

bernhard-42, 2022-03-29:
I have added a function to the next release to selectively enable/disable elements in the tollbar:

Disable:

```javascript
viewer.trimUI(["axes", "axes0", "grid", "ortho", "more"], false)
```

Enable:

```javascript
viewer.trimUI(["axes", "axes0", "grid", "ortho", "more"], true)
```

Now you can easily go down to width of 400px

[image]

Since you have to create the viewer component in the web view of VS Code, you should be able to call this function, right?

bernhard-42, 2022-03-29:
ok, for this small window the help dialog is too large :-(
So I added "help" to the function:

```javascript
viewer.trimUI(["axes", "axes0", "grid", "ortho", "more", "help"], true)
```

Now 330px is the min width:

[image]

Not sure how useful this is, however it is possible

roipoussiere, 2022-03-29:
bernhard-42:

> I'm not sure I want to spend time on this.

also bernhard-42, two hours and a few commits later:

> Now 330px is the min width

😁 Thank you so much, this will be very useful! ❤️

closed 2022-03-29 by roipoussiere

bernhard-42, 2022-03-31:
@roipoussiere I just cloned your repo, did a `yarn install`, added the trimUI line to the index html file:

[image]

and it works:

[image]

However, sometimes the cadquery view shows an empty CAD viewer on start and one has to save the python file again to see the object. And if it is empty, then trimUI didn't run

bernhard-42, 2022-03-31:
[image]

turns out in this case the `render` function is not called, only `update_view`