bernhard-42/jupyter-cadquery #79: Is it possible to run jupyter cadquerry remotely?

khoroshyy, 2022-04-10:
I try to connect from raspbery pi to a remote cadquery jupyter lab running on ubuntu 20.04. 
But I get the following error. Not sure what is wrong. Is there any way to test cadquery on remote machine without X forwarding?
Thanks.
Petro.

`InvalidArgumentsError                     Traceback (most recent call last)
Input In [13], in <cell line: 1>()
----> 1 import cadquery as cq
      2 from jupyter_cadquery import (
      3     PartGroup, Part, Edges, Faces, Vertices, show,
      4     close_viewer, close_viewers, get_viewer, open_viewer, set_defaults, get_defaults
      5 )
      7 from jupyter_cadquery.replay import replay, enable_replay, disable_replay, reset_replay

File ~/bin/miniconda39/envs/jcq3/lib/python3.8/site-packages/cadquery/__init__.py:14, in <module>
      2 from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
      3 from .occ_impl.shapes import (
      4     Shape,
      5     Vertex,
   (...)
     12     sortWiresByBuildOrder,
     13 )
---> 14 from .occ_impl import exporters
     15 from .occ_impl import importers
     17 # these items are the common implementation
     18 
     19 # the order of these matter

File ~/bin/miniconda39/envs/jcq3/lib/python3.8/site-packages/cadquery/occ_impl/exporters/__init__.py:10, in <module>
      6 from typing_extensions import Literal
      8 from OCP.VrmlAPI import VrmlAPI
---> 10 from ...cq import Workplane
     11 from ...utils import deprecate
     12 from ..shapes import Shape

File ~/bin/miniconda39/envs/jcq3/lib/python3.8/site-packages/cadquery/cq.py:61, in <module>
     54 from .utils import deprecate, deprecate_kwarg_name
     56 from .selectors import (
     57     Selector,
     58     StringSyntaxSelector,
     59 )
---> 61 from .sketch import Sketch
     63 CQObject = Union[Vector, Location, Shape, Sketch]
     64 VectorLike = Union[Tuple[float, float], Tuple[float, float, float], Vector]

File ~/bin/miniconda39/envs/jcq3/lib/python3.8/site-packages/cadquery/sketch.py:28, in <module>
     26 from .occ_impl.geom import Location, Vector
     27 from .occ_impl.importers.dxf import _importDXF
---> 28 from .occ_impl.sketch_solver import (
     29     SketchConstraintSolver,
     30     ConstraintKind,
     31     ConstraintInvariants,
     32     DOF,
     33     arc_first,
     34     arc_last,
     35     arc_point,
     36 )
     38 Modes = Literal["a", "s", "i", "c"]  # add, subtract, intersect, construct
     39 Point = Union[Vector, Tuple[Real, Real]]

File ~/bin/miniconda39/envs/jcq3/lib/python3.8/site-packages/cadquery/occ_impl/sketch_solver.py:236, in <module>
    222 # dictionary of individual constraint cost functions
    223 costs: Dict[str, Callable[..., float]] = dict(
    224     Fixed=fixed_cost,
    225     FixedPoint=fixed_point_cost,
   (...)
    232     ArcAngle=arc_angle_cost,
    233 )
--> 236 class SketchConstraintSolver(object):
    238     entities: List[DOF]
    239     constraints: List[Constraint]

File ~/bin/miniconda39/envs/jcq3/lib/python3.8/site-packages/cadquery/occ_impl/sketch_solver.py:265, in SketchConstraintSolver()
    259     # validate and transform constraints
    260 
    261     # indices of x corresponding to the entities
    262     self.ixs = [0] + list(accumulate(len(e) for e in self.entities))
    264 def _cost(
--> 265     self, x0: Array[(Any,), float]
    266 ) -> Tuple[
    267     Callable[[Array[(Any,), float]], float],
    268     Callable[[Array[(Any,), float], Array[(Any,), float]], None],
    269     Array[(Any,), float],
    270     Array[(Any,), float],
    271 ]:
    273     ixs = self.ixs
    274     constraints = self.constraints

File ~/bin/miniconda39/envs/jcq3/lib/python3.8/site-packages/nptyping/ndarray.py:73, in NDArrayMeta.__getitem__(cls, item)
     71 if not isinstance(item, tuple):
     72     raise InvalidArgumentsError(f"Unexpected argument of type {type(item)}")
---> 73 shape_expression, dtype = _get_from_tuple(item)
     74 validate_dtype(dtype)
     75 validate_shape_expression(shape_expression)

File ~/bin/miniconda39/envs/jcq3/lib/python3.8/site-packages/nptyping/ndarray.py:128, in _get_from_tuple(item)
    126     dtype = item[1]
    127 else:
--> 128     raise InvalidArgumentsError(
    129         f"Unexpected argument '{item[0]}', expecting"
    130         f" Shape[<ShapeExpression>] or"
    131         f" Literal[<ShapeExpression>]"
    132     )
    134 return shape_expression, dtype

InvalidArgumentsError: Unexpected argument '(typing.Any,)', expecting Shape[<ShapeExpression>] or Literal[<ShapeExpression>]`

bernhard-42, 2022-04-10:
@khoroshyy I had the same issue yesterday. For me it was a problem of _CadQuery_ dependencies. I had to install `nptyping=1.4.4`. The `nptyping` project had released 2.0.0
Maybe it is the same problem for you?

khoroshyy, 2022-04-11:
> 

Thanks, I will try and let people know.

closed 2022-04-11 by khoroshyy