bernhard-42/jupyter-cadquery #80: Problems with "show" in VS Code / Jupyter-Notebook

christianhauschel, 2022-05-11:
`show(obj)` works fine in VS Code and Jupyter-Notebook for a certain number of executions.

After a while, the `show` command does not display any graphics anymore. I have tried changing the "presentation" options in VS Code, without success.

Do you know what the problem could be?

christianhauschel, 2022-05-11:
[image]

christianhauschel, 2022-05-11:
Restarting the Jupyter kernel does not help... I have to restart VS Code...

bernhard-42, 2022-05-15:
@christianhauschel I tried to reproduce with the latest VS Code flipping between two different models continuously. Sometimes the communication gets slow, but it worked all the time.
Do you have a notebook where you can reproduce it?

afshawnlotfi, 2022-06-17:
Can confirm I am having the issue on the latest version of Vscode, jupyter-cadquery, and Python extension
[image]

Tried running in both ipynb and py formats. Also tried install ipykernel without any changes

bernhard-42, 2022-06-19:
@christianhauschel @afshawnlotfi 
Now I better understand the issue. I wasn't aware of ˋipywidgetsˋ being supported by VS Code in the meantime.
I can reproduce your error now, however, it seems to lie in the ˋipywidgetsˋ implementation of VS Code. 

I have no idea how to track that down at the moment and unfortunately need to flag direct support via notebooks in VS Code as currently not supported.
You might want to look at the viewer approach of Jupyter-CadQuery

closed 2025-04-16 by bernhard-42