bernhard-42/jupyter-cadquery #81: Installation file missing

Jojain, 2022-06-06:
Hello, I just tried to install jupyter cadquery via pip and I can't launch the voila viewer because of this :
`'C:\\Users\\XXX\\.jcv\\viewer.ipynb' is missing` The file isn't there, I'm in a new conda env with only CQ and JCQ installed (I installed CQ via the new pip install, I don't know if that can be a problem)

bernhard-42, 2022-06-07:
@Jojain I just tried the same on a Windows machine:

```
conda create -n jcq-3.1.0 python=3.9
conda activate jcq-3.1.0
conda install -c conda-forge -c cadquery

del .jcv\viewer.ipynb
rmdir .jcv
pip install jupyter-cadquery==3.1.0rc4

C:\Users\bernhard>jcv
[JCV] Creating a Jupyter kernel specification called 'jcv' for this conda environment
Installed kernelspec jcv in C:\ProgramData\jupyter\kernels\jcv
[JCV] Copying the voila notebook to C:\Users\bernhard\.jcv
        1 Datei(en) kopiert.
[JCV] Signing the voila notebook
Notebook already signed: C:\Users\bernhard\.jcv\viewer.ipynb
[JCV] Starting Jupyter CadQuery Viewer
...
```

Check notebook:

```
 C:\Users\bernhard>dir .jcv
 Volume in Laufwerk C: hat keine Bezeichnung.
 Volumeseriennummer: 02F5-F078

 Verzeichnis von C:\Users\bernhard\.jcv

07.06.2022  08:27    <DIR>          .
07.06.2022  08:27    <DIR>          ..
07.06.2022  08:26               685 viewer.ipynb
               1 Datei(en),            685 Bytes
               2 Verzeichnis(se), 300.797.108.224 Bytes frei
```

Are you using the right `jcv.cmd`?:

```
C:\Users\bernhard>where jcv
C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\Scripts\jcv
C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\Scripts\jcv.cmd
```

``` 
C:\Users\bernhard>type C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\Scripts\jcv.cmd
@echo off

set ZMQ_PORT=
set CAD_HEIGHT=
set CAD_WIDTH=
set THEME=light

:GETOPTS
if /I "%1" == "-p" set ZMQ_PORT=%2 & shift
if /I "%1" == "-h" set CAD_HEIGHT=%2 & shift
if /I "%1" == "-w" set CAD_WIDTH=%2 & shift
if /I "%1" == "-d" set THEME=dark
if /I "%1" == "-g" set GLASS_MODE=1
shift
if not "%1" == "" goto GETOPTS

echo [JCV] Creating a Jupyter kernel specification called 'jcv' for this conda environment
python -m ipykernel install --name jcv --display-name jcv

set JCV_PATH=%userprofile%\.jcv
set CMD=python -c "import os, jupyter_cadquery.viewer.server as c; print(os.path.dirname(c.__file__))"
for /f %%i in (' %CMD% ') do set VIEWER_PATH=%%i

echo [JCV] Copying the voila notebook to %JCV_PATH%
if not exist %JCV_PATH% mkdir %JCV_PATH%
copy %VIEWER_PATH%\viewer.ipynb %JCV_PATH%\

echo [JCV] Signing the voila notebook
jupyter trust %JCV_PATH%\viewer.ipynb

echo [JCV] Starting Jupyter CadQuery Viewer
voila --theme=%THEME% ^
    --enable_nbextensions=True ^
    --show_tracebacks=True ^
    --VoilaExecutor.kernel_name=jcv ^
    --VoilaConfiguration.file_whitelist="favicon.ico" ^
    --VoilaConfiguration.file_whitelist=".*\.js" ^
    %JCV_PATH%\viewer.ipynb
```

What does `python -c "import os, jupyter_cadquery.viewer.server as c; print(os.path.dirname(c.__file__))"` return?
This command is used to find the `viewer.ipynb` file:

```
C:\Users\bernhard>dir C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\lib\site-packages\jupyter_cadquery\viewer\v*
 Volume in Laufwerk C: hat keine Bezeichnung.
 Volumeseriennummer: 02F5-F078

 Verzeichnis von C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\lib\site-packages\jupyter_cadquery\viewer

07.06.2022  08:26               685 viewer.ipynb
               1 Datei(en),            685 Bytes
               0 Verzeichnis(se), 300.791.971.840 Bytes frei
```

bernhard-42, 2022-06-07:
I repeated the same as above with `jupyter-cadquery==3.0.0`, and again, all worked as expected.

Jojain, 2022-06-07:
So looking more closely the file isn't found so it's not copied in C:\\Users\... 
`[JCV] Copying the voila notebook to C:\Users\XXX\.jcv
Le fichier spécifié est introuvable.` -> (File cannot be found)

I am using mambaforge, the file is in the environment : `D:\Programmes\Mambaforge\envs\jcq\Lib\site-packages\jupyter_cadquery\viewer.ipynb` 

I am using the `jcv` command. So the problem is that somehow it doesn't find the file in the env ?

bernhard-42, 2022-06-07:
> Your command give me the right location, when I type it I get : D:\Programmes\Mambaforge\envs\jcq\lib\site-packages\jupyter_cadquery\viewer It doesn't return the extension though ? Can it be a problem?

This is correct, `jcv.cmd` adds the filename of the notebook after the path.

I now changed the script and let it start with `echo on` 

```
(jcq-3.1.0) C:\Users\bernhard>where jcv
C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\Scripts\jcv
C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\Scripts\jcv.cmd

(jcq-3.1.0) C:\Users\bernhard>code C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\Scripts\jcv.cmd
``` 

This is the outcome:

```
(jcq-3.1.0) C:\Users\bernhard>jcv

(jcq-3.1.0) C:\Users\bernhard>set ZMQ_PORT=

(jcq-3.1.0) C:\Users\bernhard>set CAD_HEIGHT=

(jcq-3.1.0) C:\Users\bernhard>set CAD_WIDTH=

(jcq-3.1.0) C:\Users\bernhard>set THEME=light

(jcq-3.1.0) C:\Users\bernhard>if /I "" == "-p" set ZMQ_PORT=   & shift

(jcq-3.1.0) C:\Users\bernhard>if /I "" == "-h" set CAD_HEIGHT=   & shift

(jcq-3.1.0) C:\Users\bernhard>if /I "" == "-w" set CAD_WIDTH=   & shift

(jcq-3.1.0) C:\Users\bernhard>if /I "" == "-d" set THEME=dark

(jcq-3.1.0) C:\Users\bernhard>if /I "" == "-g" set GLASS_MODE=1

(jcq-3.1.0) C:\Users\bernhard>shift

(jcq-3.1.0) C:\Users\bernhard>if not "" == "" goto GETOPTS

(jcq-3.1.0) C:\Users\bernhard>echo [JCV] Creating a Jupyter kernel specification called 'jcv' for this conda environment
[JCV] Creating a Jupyter kernel specification called 'jcv' for this conda environment

(jcq-3.1.0) C:\Users\bernhard>python -m ipykernel install --name jcv --display-name jcv
Installed kernelspec jcv in C:\ProgramData\jupyter\kernels\jcv

(jcq-3.1.0) C:\Users\bernhard>set JCV_PATH=C:\Users\bernhard\.jcv

(jcq-3.1.0) C:\Users\bernhard>set CMD=python -c "import os, jupyter_cadquery.viewer.server as c; print(os.path.dirname(c.__file__))"

(jcq-3.1.0) C:\Users\bernhard>for /F %i in (' python -c "import os, jupyter_cadquery.viewer.server as c; print(os.path.dirname(c.__file__))" ') do set VIEWER_PATH=%i

(jcq-3.1.0) C:\Users\bernhard>set VIEWER_PATH=C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\lib\site-packages\jupyter_cadquery\viewer

(jcq-3.1.0) C:\Users\bernhard>echo [JCV] Copying the voila notebook to C:\Users\bernhard\.jcv
[JCV] Copying the voila notebook to C:\Users\bernhard\.jcv

(jcq-3.1.0) C:\Users\bernhard>if not exist C:\Users\bernhard\.jcv mkdir C:\Users\bernhard\.jcv

(jcq-3.1.0) C:\Users\bernhard>copy C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\lib\site-packages\jupyter_cadquery\viewer\viewer.ipynb C:\Users\bernhard\.jcv\
        1 Datei(en) kopiert.

(jcq-3.1.0) C:\Users\bernhard>echo [JCV] Signing the voila notebook
[JCV] Signing the voila notebook

(jcq-3.1.0) C:\Users\bernhard>jupyter trust C:\Users\bernhard\.jcv\viewer.ipynb
Notebook already signed: C:\Users\bernhard\.jcv\viewer.ipynb

(jcq-3.1.0) C:\Users\bernhard>echo [JCV] Starting Jupyter CadQuery Viewer
[JCV] Starting Jupyter CadQuery Viewer

(jcq-3.1.0) C:\Users\bernhard>voila --theme=light     --enable_nbextensions=True     --show_tracebacks=True     --VoilaExecutor.kernel_name=jcv     --VoilaConfiguration.file_whitelist="favicon.ico"     --VoilaConfiguration.file_whitelist=".*\.js"     C:\Users\bernhard\.jcv\viewer.ipynb
[Voila] Using C:\Users\bernhard\AppData\Local\Temp to store connection files
[Voila] Storing connection files in C:\Users\bernhard\AppData\Local\Temp\voila_ke7ah2nr.
[Voila] Serving static files from C:\Users\bernhard\miniconda3\envs\jcq-3.1.0\lib\site-packages\voila\static.
[Voila] Voilà is running at:
http://localhost:8866/
[Voila] Kernel started: 9e63956b-203f-466b-a5d2-83212654d8cb
WARNING:tornado.access:404 GET /voila/files/favicon.ico (::1) 0.00ms
```

Maybe you could do the same and past the result?

Jojain, 2022-06-07:
Running the script with echo on : 
```
(base) PS D:\Programmes\Mambaforge> jcv -d

D:\Programmes\Mambaforge>set ZMQ_PORT=

D:\Programmes\Mambaforge>set CAD_HEIGHT=

D:\Programmes\Mambaforge>set CAD_WIDTH=

D:\Programmes\Mambaforge>set THEME=light

D:\Programmes\Mambaforge>if /I "-d" == "-p" set ZMQ_PORT=   & shift

D:\Programmes\Mambaforge>if /I "-d" == "-h" set CAD_HEIGHT=   & shift

D:\Programmes\Mambaforge>if /I "-d" == "-w" set CAD_WIDTH=   & shift

D:\Programmes\Mambaforge>if /I "-d" == "-d" set THEME=dark

D:\Programmes\Mambaforge>shift

D:\Programmes\Mambaforge>if not "" == "" goto GETOPTS

D:\Programmes\Mambaforge>echo [JCV] Creating a Jupyter kernel specification called 'jcv' for this conda environment
[JCV] Creating a Jupyter kernel specification called 'jcv' for this conda environment

D:\Programmes\Mambaforge>python -m ipykernel install --name jcv --display-name jcv
Installed kernelspec jcv in C:\ProgramData\jupyter\kernels\jcv

D:\Programmes\Mambaforge>set JCV_PATH=C:\Users\romain.ferru\.jcv

D:\Programmes\Mambaforge>set CMD=python -c "import os, jupyter_cadquery.viewer.server as c; print(os.path.dirname(c.__file__))"

D:\Programmes\Mambaforge>for /F %i in (' python -c "import os, jupyter_cadquery.viewer.server as c; print(os.path.dirname(c.__file__))" ') do set VIEWER_PATH=%i
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "D:\Programmes\Mambaforge\lib\site-packages\jupyter_cadquery\__init__.py", line 33, in <module>
    from .cad_objects import (
  File "D:\Programmes\Mambaforge\lib\site-packages\jupyter_cadquery\cad_objects.py", line 24, in <module>
    from cadquery.occ_impl.shapes import Face, Edge, Wire
ModuleNotFoundError: No module named 'cadquery'

D:\Programmes\Mambaforge>echo [JCV] Copying the voila notebook to C:\Users\XXX\.jcv
[JCV] Copying the voila notebook to C:\Users\romain.ferru\.jcv

D:\Programmes\Mambaforge>if not exist C:\Users\XXX\.jcv mkdir C:\Users\romain.ferru\.jcv

D:\Programmes\Mambaforge>copy \viewer.ipynb C:\Users\XXX\.jcv\
Le fichier spécifié est introuvable.
.
.
.

```
So the problem seems to come from the fact that it cannot import cadquery (which is installed in these envs).
I tried with Miniconda3 and it works. I assume there is a problem with mambaforge envs, I don't mind it I'm going to use miniconda instead of mambaforge and assume the bug is on their side. Sorry for the inconvenience

closed 2022-06-07 by Jojain