bernhard-42/jupyter-cadquery #82: can not get ocp version

SubChange, 2022-06-27:
I have installed package cadquery and jupyter-cadquery through following commands successfully:

```text
conda install -c cadquery cadquery
conda install -c cadquery ocp
```
however, when I running the given examples in `example/1-cadquery.ipynb`, An error occurred

[image]

It seems that I have no file in the corresponding path `f"{os.environ['CONDA_PREFIX']}/lib/libTKBRep.*.*.*"` - function `ocp_version` in `ocp_utils.py` , what I have in folder `envs/env_name/Lib/site-packages` are as follows:

[image]

Is there any updates for this function?

bernhard-42, 2022-07-03:
libTK... libraries are not from OCP but from occt and occt does not install into site-packages, but into '$CONDA_prefix/lib'. Is occt properly installed for you?

bernhard-42, 2022-07-10:
Not sure what happened here, but to be on the safe side, 5e latest version 3.1 catches this exception

bernhard-42, 2022-08-02:
fixed in 3.1. Will close. If you still find issues, please reopen

closed 2022-08-02 by bernhard-42