bernhard-42/jupyter-cadquery #83: Allow setting "up"

tjltjl, 2022-07-26:
First, thank you for an amazing piece of software!

I would really love to be able to specify what vector should be "up" in the viewer to be able to easily view models in the more standard CAD orientation of "X right, Y up, Z towards viewer".

I'm guessing this cuts through all three layers since the "up" vector is set at the lowest level. And it is fixed to be the Z axis so for models oriented in the standard CAD way, this gets messy fast.

I used to hate that coordinate system but little by little, using e.g. fusion 360 gets you used to it.

The easy workaround is of course to rotate the model but then, editing the model and looking at the coordinate axes shown on the screen gets really confusing.

bernhard-42, 2022-07-31:
@tjltjl I installed Fusion 360 to better understand.
Looks like the changes so that Jupyterlab behaves like Fusion 360 would just be in the lowest javascript layer.
I'll give it a try - no promise ;-)

tjltjl, 2022-07-31:
Having the option of Z being up (as currently) or Y being up (Fusion) would be really nice

bernhard-42, 2022-08-02:
@tjltjl Would you have time to quickly check out https://bernhard-42.github.io/three-cad-viewer/example.html and select the Example `dirbox` and comare the Controls `trackball/Y up` and `trackball/Z up` (+X, +Y, +Z are faces in positive axis direction and -X, -Y, -Z in negative)?
Is `trackball/Y up` as you would expect it?

[image]

tjltjl, 2022-08-04:
Perfect!

bernhard-42, 2022-08-13:
@tjltjl 
I've release version 3.2 which supports a parameter `up='Y'` for `show`. Default is `up='Z'`

bernhard-42, 2022-09-18:
Closing this since it is supported since 3.2
If there is still an issue, please feel free to re-open

closed 2022-09-18 by bernhard-42