bernhard-42/jupyter-cadquery #85: 3.2.0 Release on PyPI doesn't contain latest change

wzhd, 2022-08-19:
On Aug 13, the `__init__` method of `Color` was changed to set the alpha first:

https://github.com/bernhard-42/jupyter-cadquery/blob/9e0a093a5ab0610cad852b9414c93ccf8a032da8/jupyter_cadquery/utils.py#L15 

But download and extract the [PyPI version](https://pypi.org/project/jupyter-cadquery/3.2.0/#files), in `jupyter_cadquery/utils.py`, the 15th line is  `if color is None:`

Users may encounter a problem like `AttributeError: 'Color' object has no attribute 'a'` when trying to use `show`.

bernhard-42, 2022-08-20:
Thanks @wzhd for finding it. Some error seems to have happened during deployment. I will deploy a version 3.2.1 to ensure consistency

bernhard-42, 2022-08-20:
@wzhd 
arrrgh, two days ago ipywidgets published version 8.0
I have tagged 7.7 in `cad-viewer-widget`, however, the latest jupyterlab 3.4.5 installs the newer javascript widget library, which leads to a failure of Jupyter CadQuery. It does work with Jupyterlab 3.4.3.

I need to further investigate, how to deal with this (supporting Jupyterlab up to 3.4.3 and above 3.4.5)

bernhard-42, 2022-08-20:
@wzhd If you want to run Jupyterlab 3.4.5, please install it via `pip install jupyterlab jupyterlab-widgets==1.1.1` 
This will ensure ipywidgets 7.7 to be correctly supported

bernhard-42, 2022-08-21:
3.2.2 now uses cad-viewer-widget 1.3.5 which fixes ipywidgets version 7.7.2

bernhard-42, 2022-09-18:
Closing this for now since 3.2.2 fixed the issue
If you still have problems, please feel free to re-open

closed 2022-09-18 by bernhard-42